daniestevez/gr-satellites #400: Error when building main branch in Ubuntu 22.04

astrohead, 2022-08-30:
After cloning the main branch of the repository and executing `cmake..` on the` /build` directory I get the following error message:

```
CMake Error at /usr/lib/cmake/pybind11/pybind11Tools.cmake:165 (add_library):
  Target "satellites_python" links to target "sndfile::sndfile" but the                                                                                                                      
  target was not found.  Perhaps a find_package() call is missing for an                                                                                                                     
  IMPORTED target, or an ALIAS target is missing?                                                                                                                                            
Call Stack (most recent call first):                                                                                                                                                         
  /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrPybind.cmake:261 (pybind11_add_module)                                                                                                          
  python/bindings/CMakeLists.txt:62 (GR_PYBIND_MAKE_OOT)                                                                                                                                     
                                                                                                                                                                                             
                                                                                                                                                                                             
CMake Error at lib/CMakeLists.txt:98 (add_library):
  Target "gnuradio-satellites" links to target "sndfile::sndfile" but the                                                                                                                    
  target was not found.  Perhaps a find_package() call is missing for an                                                                                                                     
  IMPORTED target, or an ALIAS target is missing?   
```   

Any idea on what/where the issue could be?

Thanks

daniestevez, 2022-08-30:
You need to install the package `libsndfile1-dev`. I now realize that this isn't said in the documentation, so I'll update it.

daniestevez, 2022-08-30:
Documentation updated in 4aaa916f4cf69e1c7f119a1360a268d6729370ae

astrohead, 2022-08-30:
Thanks a lot for your fast reply @daniestevez !

After installing the package `libsndfile1-dev` I could succesfully install `gr-satellites` and use it with GNURadio 3.10 + Ubuntu 22.04

daniestevez, 2022-08-30:
Great. Closing the issue.

closed 2022-08-30 by daniestevez