daniestevez/gr-satellites #404: Usage of placebo set_min_noutput_items()

willcode, 2022-09-19:
The `set_min_noutput_items()` function in GNU Radio has no effect, as the affected `d_min_noutput_items` is never used internally.

gr-satellites uses this function in:
- lib/varlen_packet_framer_impl.cc
- lib/varlen_packet_tagger_impl.cc

This may not be causing any probems, but might be worth a check.

willcode, 2022-09-19:
Nevermind :-)

closed 2022-09-19 by willcode

daniestevez, 2022-09-20:
Thanks for spotting this. These two blocks are something that Glenn Richardson contributed quite a while ago. I'm not so familiar with them and I'm not even sure I've ever used them. They are/were part of a setup to transmit AX100 frames (while most of gr-satellites focuses on receiving).

I'm not sure what to do with these two blocks lying around, as I don't know if anyone actually uses them. It might happen that there are better ways to do the same with gr-pdu blocks, but I can't recall what these blocks do exactly.

willcode, 2022-09-21:
A few minutes after putting in Issues on a couple of projects, I double checked the GR source code and figured out I was wrong, so you can ignore this.