daniestevez/gr-satellites #407: qa_phase_unwrap test fails

mripv6, 2022-10-10:
I stumbled around trying to install gr-satellites using the PPA and wasn't able to make gr_satellites work. That lead me to attempt to build from the sources, but the tests failed. After reading the instructions again, I figured out volk_profile was never run. PHYTHONPATH wasn't set. After making these two fixes, the package from the PPA worked. Perhaps you may want to add a notice about these items in the Ubuntu PPA installation instructions. (Using recently upgraded Linux Mint version 21.)

After making the two changes, I also compiled from sources and ran 'make test'. All the tests except the qa_phase_unwrap_test passed. Here's the output from LastTest.log. Incidentally, you may want to suggest to use 'make -j4' in the compile instructions. You may also want to suggest to use the --install-prefix switch on cmake to put the files in /usr/lib instead of the default /usr/local/lib. Since I was able to get gr_satellites working with the one installed from the PPA AND all the tests didn't pass, I didn't do a 'make install'. 

Hope this helps.

2/20 Testing: qa_phase_unwrap
12/20 Test: qa_phase_unwrap
Command: "/usr/bin/sh" "qa_phase_unwrap_test.sh"
Directory: /data/home/stu/gr-satellites/build/python
"qa_phase_unwrap" start time: Oct 10 15:29 EDT
Output:
----------------------------------------------------------
Traceback (most recent call last):
  File "/data/home/stu/gr-satellites/python/qa_phase_unwrap.py", line 25, in <module>
    from satellites import phase_unwrap
ImportError: cannot import name 'phase_unwrap' from 'satellites' (/usr/lib/python3/dist-packages/satellites/__init__.py)
<end of output>
Test time =   0.25 sec
----------------------------------------------------------
Test Failed.
"qa_phase_unwrap" end time: Oct 10 15:29 EDT
"qa_phase_unwrap" time elapsed: 00:00:00
----------------------------------------------------------

daniestevez, 2022-10-11:
I have just tested installing gr-satellites from the PPA in a fresh Ubuntu (using the Dockerfile in the main branch of [gr-satellites-ci-action](https://github.com/daniestevez/gr-satellites-ci-action)). It has just worked without the need to do anything special.

When gr-satellites is installed from the PPA package, it goes into `/usr/lib/python3/dist-packages`, so it is not necessary to set the `PYTHONPATH`. Additionally, running `volk_profile` is not necessary.

Regarding why the tests fail, it seems that you have still the gr-satellites from the PPA installed in your system (which is v5.0.0), and are running the tests from the main branch. The test is testing against the PPA installation, which doesn't have the phase_unwrap block, because it was added after v5.0.0.

> You may also want to suggest to use the --install-prefix switch on cmake to put the files in /usr/lib instead of the default /usr/local/lib

What is wrong with installing to `/usr/local/lib` by default?

mripv6, 2022-10-12:
Thanks for the reply. At this point, I've tried so many different things 
that I'm sure I have too many artifacts hanging around.

I'll try removing the PPA and do clean rebuild with an install. 
Although, it would seem the tests would use the code in the build 
directory and not use something installed elsewhere on the system.

On Linux Mint, all the python3 packages seem to be installed in 
/usr/lib/python3/dist-packages. satellites was installed in 
/usr/local/lib/python3.10/dist-packages and it was the only package 
there. So my thought was to build satellites with the 'standard' python 
dist-packages prefix for my system.

Thanks for your help.

daniestevez, 2022-10-12:
You're right that when running the tests by doing `make test` or `ctest` they should run against whatever is in `build` rather than whatever has been installed system-wide. However, the tests are also set up such that when run as a standalone Python script, the tests should test what is installed system-wide.

Getting these two behaviours is somewhat tricky, and it is done [like so](https://github.com/daniestevez/gr-satellites/blob/main/python/qa_phase_unwrap.py#L16). This is not perfect, because if the Python module hasn't been built properly, in the sense that `import python as satellites` from the build directory raises an `ImportError`, but there is a system-wide installation that can be imported as `import satellites`, then `make test` will run against such an installation.

I'm open to suggestions about how to restructure this.

daniestevez, 2022-10-20:
Can this issue be closed?

closed 2022-10-20 by daniestevez