daniestevez/gr-satellites #413: Swig error

sv1, 2022-10-26:
I am trying to upgrade from 3.14 to 3.15 but the build stops at 96%

error: cannot convert ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’} to ‘const char*’
36337 |       result = gr::satellites::doppler_correction::make(*arg1,arg2,arg3);
      |                                                         ^~~~~
      |                                                         |
      |                                                         std::string {aka std::__cxx11::basic_string<char>}

daniestevez, 2022-10-26:
Can you give more information about your system and include the full build log?

sv1, 2022-10-27:
On Wed, 2022-10-26 at 13:57 -0700, Daniel Estévez wrote:
> Can you give more information about your system and include the full build
> log?
> 
Hello Daniel,

I am trying to build 3.15.0 on Debian 11.5. attached are the logfiles.

73
Apostolos

daniestevez, 2022-10-27:
Thank you. It seems that this is related to #398, in which having a `std::string&` parameter in the constructor of the doppler_correction block was giving problems with SWIG. I changed this to `const char*` in 03cdc4f3f4b8f84baff7cb67d1bd6af776841baf and it fixed the problem.

But somehow this doesn't work with your system, because it's attempting to call `doppler_correction::make` with a `std::string` rather than a `char *`.

The CI is passing with Ubuntu 20.04 and the GNU Radio 3.8 from that distribution, so I wonder what's different in your system.

Can you try to revert 03cdc4f3f4b8f84baff7cb67d1bd6af776841baf and see if that fixes the problem in your system?

daniestevez, 2022-10-27:
The following Dockerfile is able to build gr-satellites v3.15.0 correctly under Debian 11.5, so I think that the problem is something about your system. Can you spot anything where your system setup deviates from what I'm doing in the Dockerfile?

```Dockerfile
# syntax=docker/dockerfile:1
FROM debian:11.5
RUN apt-get update \
    && DEBIAN_FRONTEND="noninteractive" apt-get -y upgrade \
    && DEBIAN_FRONTEND="noninteractive" apt-get install -y \
           gnuradio build-essential cmake git swig liborc-0.4-dev \
	   python3-pip \
    && pip3 install construct \
    && git clone https://github.com/daniestevez/gr-satellites \
    && cd gr-satellites \
    && git checkout v3.15.0 \
    && mkdir build \
    && cd build \
    && cmake .. \
    && make -j$(nproc) \
    && make test \
    && make install
```

sv1, 2022-10-27:
I can see the following diffirences:

I have cmake 3.24.2 instead of 3.18.4,
	git 2.34.1 instead of 2.30.2
and 	python3-construct 2.10.58 from debian repos instead of pip3

sv1, 2022-10-27:
Yes! reverting the commit gr-satellites build successfully!

daniestevez, 2022-10-27:
I wonder why you have different versions of cmake and git, but these wouldn't explain the problem.

Good that reverting the commit fixes the problem for you, but I don't think it's possible to revert it in the repo, since it might break in other people's systems.

daniestevez, 2023-09-26:
Closing this, since it is quite old and perhaps is no longer applicable. Feel free to reopen if a similar problem appears.

closed 2023-09-26 by daniestevez