daniestevez/gr-satellites #424: Throttle not working when wavfile is used

kng, 2022-12-01:
Wavfile can get the `--samp_rate` from the audio file, but doesn't seem to be available for `--throttle`.

```
$ gr_satellites 53385 --wavfile satnogs_6814977_2022-11-30T15-33-45.ogg --throttle
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 328, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 313, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 212, in __init__
    self.throttle = blocks.throttle(size,
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gnuradio.blocks.blocks_python.throttle(itemsize: int, samples_per_sec: float, ignore_tags: bool = True)

Invoked with: 4, None, True
```

Tested on gr_satellites v5.1.0-git gr_satellites v5.1.1

daniestevez, 2022-12-01:
Thanks. This is definitely a bug.

I've taken a look and the fix was easy. It should be fixed in e8bcd1bdf6e0ebd304d3674d8bce11e968ab51c7.

daniestevez, 2022-12-09:
Can this issue be closed, given that it's fixed in https://github.com/daniestevez/gr-satellites/commit/e8bcd1bdf6e0ebd304d3674d8bce11e968ab51c7?

kng, 2022-12-09:
happy times (:

closed 2022-12-09 by kng