daniestevez/gr-satellites #425: Missing tags in "Fixed Length Packet Tagger"

notsleep196, 2022-12-02:
Hello!
I faced the issue with tags propagation in "Fixed Length Packet Tagger" block. I tried to push to flowgraph some additional info via tags (using PDU) and saw that this tags are mixed or missed after "Fixed Length Packet Tagger". So i made easier flowgraphs to demonstrate this.

My TopBlock looks like this: 
[image]
gr_satellites_bug_hier looks like this:
[image]
As you can see from TopBlock main idea is to decode data with different "syncwords" in one flowgraph. 

I swapped PadSource with simple custom MessagePublisher to push messages to flowgraph. The final python file of TopBlock:

<details>
  <summary>TopBlock</summary>

```python
import os
import sys
import pmt

from gnuradio import blocks
from gnuradio import digital
from gnuradio import gr
from gnuradio.filter import firdes
from gnuradio.fft import window
import signal
from argparse import ArgumentParser
from gnuradio.eng_arg import eng_float, intx
from gnuradio import eng_notation
from gnuradio import gr, pdu
from gr_satellites_bug_hier import gr_satellites_bug_hier  # grc-generated hier_block

class MessagePublisher(gr.sync_block):

    def __init__(self):
        gr.sync_block.__init__(
            self,
            name="Publisher",
            in_sig=None,
            out_sig=None
        )
        self.message_port_register_out(pmt.intern('out'))

class gr_satellites_bug(gr.top_block):

    def __init__(self):
        gr.top_block.__init__(self, "gr-satellites_bug_TopBlock", catch_exceptions=True)

        ##################################################
        # Variables
        ##################################################
        self.syncword_small_packet = syncword_small_packet = "0111000110011101100000111100100101010011010000100010110111111010"
        self.syncword_long_packet = syncword_long_packet = "0010010011001000110101101001110000000110000101110111100010101111"
        self.mtu_small_packet = mtu_small_packet = 32
        self.mtu_long_packet = mtu_long_packet = 64
        self.message_publisher = MessagePublisher()
        self.iter = 1

        ##################################################
        # Blocks
        ##################################################
        self.pdu_pdu_to_tagged_stream_0 = pdu.pdu_to_tagged_stream(gr.types.byte_t, 'length')
        self.gr_satellites_bug_hier_0_0 = gr_satellites_bug_hier(
            mtu=mtu_small_packet,
            syncword=syncword_small_packet,
        )
        self.gr_satellites_bug_hier_0 = gr_satellites_bug_hier(
            mtu=mtu_long_packet,
            syncword=syncword_long_packet,
        )
        self.digital_chunks_to_symbols_xx_0 = digital.chunks_to_symbols_bf([-1,1], 1)
[... 386 lines cut ...]
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0150: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0160: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0170: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
01a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
01b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
01c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
************************************
correlate_access_code_tag_ff :debug: writing tag at sample 16160

</blink>
</details>

As i can see from Tag Debug info, tags in "inactive" (e.g. "syncword" not detected) part of flowgraph are buffered in "Fixed Length Packet Tagger" block. When pushing to TopBlock frames with different "syncword" block works unexpectedly, mixing and, sometimes, missing all additional tags.

Environment:
GNU Radio: 3.10.1.1
gr-satellites: 5.0.0
python: 3.10.6

daniestevez, 2022-12-02:
Hi, thanks for reporting this.

You are right. `fixedlen_tagger` doesn't propagate tags from the input to the output. The code to do that is simply missing. Additionally, `fixedlen_tagger` (which is a Python block, and hence not so efficient) was mostly deprecated in May this year in favour of the `fixedlen_to_pdu` block (which is a C++ block and much more efficient). I see that I forgot to mark the GRC block as deprecated, so maybe I should do that.

`fixedlen_to_pdu` is somewhat different to `fixedlen_tagger`. Its output are PDUs rather than a tagged stream. This made more sense, because gr-satellites always converts the tagged stream to PDUs, and so it used Tagged Stream to PDU always after `fixedlen_tagger`. Since it's more difficult to properly format a tagged stream output, it makes more sense to have a block which outputs PDUs directly.

`fixedlen_to_pdu` doesn't propagate tags from input to output either. We could add code so that the tags get added to the PDU, but currently there is no need for this in gr-satellites.

I see that the flowgraph you have could likely be achieved by using gr-satellites Sync and Create Packed PDU block. But perhaps this is just an example to demonstrate the problem rather than your final goal.

In any case, if you do need to use `fixedlen_tagger` in your application and want to implement tag propagation yourself, I'm happy to review the code and get it merged upstream. At the moment, there is not much interest in having tag propagation in these blocks, so it's probably not something that I'll implement myself.

notsleep196, 2023-01-09:
Hello and thank you for reply!
I think i can close this issue, because your answer is exhaustive.

daniestevez, 2023-01-09:
Thanks. I actually prefer to leave this issue open, since it describes a possible enhancement that someone might want to implement in the future.