daniestevez/gr-satellites #428: gr_satellites on ogg file never finishes

kng, 2022-12-11:
I stumbled on some ogg files that would never complete and exit when I processed a bunch of observations. 
The audio from [this observation](https://network.satnogs.org/observations/6857031/) run with `gr_satellites 53385 --wavfile satnogs_6857031_2022-12-10T09-43-12.ogg`.
But converting it to wav with sox makes it run to completion and exit.
`sox satnogs_6857031_2022-12-10T09-43-12.ogg satnogs_6857031_2022-12-10T09-43-12.wav`
`gr_satellites 53385 --wavfile satnogs_6857031_2022-12-10T09-43-12.wav`
This points to libsndfile or something else ?
Tried running with `--dump_path` and the files didn't continue to grow.
Tested with gr_satellites version v5.2.0-git and v5.1.0-git
```
ii  libsndfile1:amd64     1.1.0-3+b1   amd64        Library for reading/writing audio files
ii  libsndfile1-dev:amd64 1.1.0-3+b1   amd64        Development files for libsndfile; a library for reading/writing audio files
ii  python3        3.10.6-1     amd64        interactive high-level object-oriented language (default python3 version)
ii  liborc-0.4-0:amd64           1:0.4.32-2   amd64        Library of Optimized Inner Loops Runtime Compiler
ii  liborc-0.4-dev:amd64         1:0.4.32-2   amd64        Library of Optimized Inner Loops Runtime Compiler (development headers)
ii  libspdlog-dev:amd64 1:1.10.0+ds-0.4 amd64        Very fast, header only or compiled, C++ logging library
ii  libspdlog1.10:amd64 1:1.10.0+ds-0.4 amd64        Very fast C++ logging library
```

daniestevez, 2022-12-12:
Hi,
I have been able to reproduce this bug with a flowgraph that only has a Wav File Source followed by a Null Sink:
[test_ogg.grc.zip](https://github.com/daniestevez/gr-satellites/files/10211640/test_ogg.grc.zip)
This flowgraph doesn't terminate either.

So this seems like a GNU Radio bug. Maybe this particular OGG file is slightly malformed or has some other weirdness that makes the Wav File Source block never return `WORK_DONE` (which is what would trigger the flowgraph to terminate). With other OGG files, gr-satellites does terminate properly, so I guess this simple test flowgraph also would terminate.

I suggest you [open up a bug report for GNU Radio](https://github.com/gnuradio/gnuradio/issues), since there is probably not much that can be done in gr-satellites to fix this.

kng, 2022-12-14:
Bug report opened.

closed 2022-12-14 by kng