daniestevez/gr-satellites #435: Error with gr-satellites

lfainoveng, 2023-02-03:
I was using the RMS AGC block in gnuradio in a functional flowgraph working 6 months ago.

Yesterday i started the flowgraph and produced this error in the flowgraph(and then running just gr_satellites by cmd line):

$ gr_satellites
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/satellites/__init__.py", line 42, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 16, in <module>
    import satellites.core
  File "/usr/local/lib/python3.8/dist-packages/satellites/__init__.py", line 44, in <module>
    from .satellites_python import *
ImportError: libgnuradio-runtime.so.3.9.5: cannot open shared object file: No such file or directory

What could possible happened in updates is gnuradio jumped to 3.10 version in a regular update and the gr-satellites was installed trough ppa so it could be a mismatch in versions... But producing error in command line calling gr_satellites makes me think that the problem could be some import that is no longer available.

Already downgraded to gnuradio 3.8 and still the same error.

Hope to find a solution with your help.

Thank you

daniestevez, 2023-02-03:
Hi,
I think you're using a gr-satellites installation that you built from source against GNU Radio 3.9. The Ubuntu PPA packages get installed to `/usr/` instead of `/usr/local/`. It might happen that you have several "conflicting" installations of gr-satellites in your system.

I suggest you clean up your old installations first. Then, if you're using Ubuntu 22.04 or 22.10 you can use the PPA package together with the Ubuntu GNU Radio package (not the GNU Radio PPA package, since gr-satellites is built against Ubuntu's GNU Radio version). If not, you can build and install gr-satellites from source.

daniestevez, 2023-04-28:
Have you tried any of the suggested solutions?

lfainoveng, 2023-05-02:
Yeah and solved the issue, thank you for your precious support

daniestevez, 2023-05-02:
Great. Closing.

closed 2023-05-02 by daniestevez