daniestevez/gr-satellites #478: No Decoding

ik8ozv, 2023-07-09:
Hello Folks,
please I am trying to decode the satellite but I do not get any incoming datas.
At the bottom window of the gnuradio I can just read:

correlate_access_code_tag_bb :debug: writing tag at sample 10386
correlate_access_code_tag_bb :debug: writing tag at sample 34649
correlate_access_code_tag_bb :debug: writing tag at sample 4754
correlate_access_code_tag_bb :debug: writing tag at sample 29580
correlate_access_code_tag_bb :debug: writing tag at sample 64486
correlate_access_code_tag_bb :debug: writing tag at sample 122344
correlate_access_code_tag_bb :debug: writing tag at sample 69761
correlate_access_code_tag_bb :debug: writing tag at sample 17948

Please where am I still missing?
Any Help?
Thanks
Enzo

daniestevez, 2023-07-09:
Hello Enzo, can you give more information about what you're doing? If you tell us exactly how you're running gr-satellites and share a recording we will be able to replicate your problem.

ik8ozv, 2023-07-09:
Thanks Dani,
for coming back, As i have lost all my datas due to an HD failure i am trying to put on again all.
I have installed all but when i run a grc flowgraps i got the issue.
Please what do k you need to know that vìcan help yopu to understand where am i missing?
Thanks

ik8ozv, 2023-07-13:
@daniestevez
Dani please any news for me?
Thanks
Enzo

daniestevez, 2023-07-13:
With the information you have given it is really difficult to help you.

Please indicate what you are doing with as much detail as possible, including, but not limited to: what version of gr-satellites you are running, what is your operating system, where did you get that `IRIS_A.py` (since such a file is not distributed with gr-satellites anymore). Also, since the problem could be with your SDR receiver, make a recording, try to decode from the recording, and share the recording.

ik8ozv, 2023-07-16:
Dani,
i have finally solved the problem with the FCDPP but now i am looking for some flowgraphs to decode the satellite with the FCDPP. Please can you help me?
I ma using Ubuntu 22.04.02 with gnuradio 3.10.1.1.
I have tried to open the ones in gr-satellites/examples/satellites
and other but i get them with red blocks and also i do not see the FCDPP as source.
Maybe before i lost all something has changed.
Please can you point me to the correct way?
Thanks in advance

73's de Enzo  IK8OZV

daniestevez, 2023-07-16:
The easiest way to set this up is probably to use an external application such as GQRX to control the FCDPP, tune to the signal of interest and stream audio data by UDP, and then use the `gr_satellites` command line tool to receive the UDP data and perform decoding. The [documentation](https://gr-satellites.readthedocs.io/en/latest/command_line.html#gr-satellites-command-line-tool) for the command line tool can be helpful.

ik8ozv, 2023-07-16:
Dani,
right now i had a pass of Gomx-1 and following your suggestion i have 
ran in the terminal the comand:

gr_satellites gomx-1 --udp --samp_rate 48e3

using the GQRX port 7355 and have noticed that when I got a packet I got in the terminal:

correlate_access_code_tag_bb :debug: writing tag at sample 32762
correlate_access_code_tag_bb :debug: writing tag at sample 36978
correlate_access_code_tag_bb :debug: writing tag at sample 118515

Please any suggestions?
Thanks

daniestevez, 2023-07-16:
Maybe the only problem is that the signal quality is not good enough. I suggest you try doing an IQ recording first, and then work with that IQ recording, and share the recording, so that others can try to replicate what you're doing.

ik8ozv, 2023-07-23:
Dani Followed your suggestion. Thanks.
This is what i have done on an IRIS-A pass.
I ran in the terminal the comand:

gr_satellites IRIS-A --udp --samp_rate 48e3

and i got as response:

udp_source :info: Listening for data on UDP port 7355.

I have opened GQRX and enabled the UDP button after been sure that the network port was set as 7355.
After a couple of burst i did not get anything on the terminal so i have , as you suggested me, enabled 
the REC button and continued the pass.
After the pass i have loaded in gnuradio the SATELLITE_DECODER.grc from the folder:

/gr-satellites/examples/satellite_decoder

I have the chenged the Satellite Decoder block changing the Satellite name to IRIS-A
and the Wav file source block pointing o the recorded file.
Ran it but did nit see anythin so i have tiefe to switch from NO to YES the Repeat field in the Wav File Source block.
Ran again and this time after a couple of seconds got:

message_debug :info: The `print_pdu` port is deprecated and will be removed; forwarding to `print`.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 30 bytes
pdu vector contents = 
0000: df b6 f7 1f 67 0c 5d 03 1f 80 62 89 5c 33 1d 05 
0010: 96 b6 29 84 e3 30 3d fb bf a0 19 f2 4c e0 
************************************
message_debug :info: The `print_pdu` port is deprecated and will be removed; forwarding to `print`.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 32 bytes
pdu vector contents = 
0000: 00 23 6d cd 4a b7 9c 9e 13 12 8e 1a b8 75 94 67 
0010: a3 f6 2f 68 89 49 11 2d 79 77 6f b9 33 66 a4 d8 
************************************
message_debug :info: The `print_pdu` port is deprecated and will be removed; forwarding to `print`.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 16 bytes
pdu vector contents = 
0000: a0 46 65 f5 3c c6 6c 9e b7 16 25 d5 ca b6 50 ad 
************************************

Please why i do not get anything when i run in the terminal:

gr_satellites IRIS-A --udp --samp_rate 48e3

Thanks.
P.S. if you need the recorded file please tell me how to upload it.
Thanks a lot

daniestevez, 2023-07-23:
From a quick look at gqrx_20230723_103610_436920600.zip, the frequency tuning seems to be off by a lot (since there is a large DC offset in the FM demodulated signal), and the file is clipped (probably because the output gain in GQRX is set too high).

ik8ozv, 2023-07-23:
Now i am receiving the Greencube and made the same steps and get:

correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: Access code: 930b51de
correlate_access_code_tag_bb :debug: Mask: ffffffff
correlate_access_code_tag_bb :debug: writing tag at sample 109470
correlate_access_code_tag_bb :debug: writing tag at sample 218211
correlate_access_code_tag_bb :debug: writing tag at sample 785998
correlate_access_code_tag_bb :debug: writing tag at sample 914208
correlate_access_code_tag_bb :debug: writing tag at sample 915488
correlate_access_code_tag_bb :debug: writing tag at sample 983684
correlate_access_code_tag_bb :debug: writing tag at sample 1185213
correlate_access_code_tag_bb :debug: writing tag at sample 1252364
correlate_access_code_tag_bb :debug: writing tag at sample 722240
correlate_access_code_tag_bb :debug: writing tag at sample 789445
correlate_access_code_tag_bb :debug: writing tag at sample 1856809
correlate_access_code_tag_bb :debug: writing tag at sample 957458

Thanks

ik8ozv, 2023-07-29:
Hello Dani,
Please any news?
I am still struggling to understand how the demodulate and decode the satellites in real time.
Any help?
Thanks in advance
73’s de Enzo IK8OZV