daniestevez/gr-satellites #479: No module named 'satellites.bindings'

RubenMardonesVarela, 2023-07-10:
I cannot run gr_satellites as I always get this module error: 

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/satellites/__init__.py", line 47, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 20, in <module>
    import satellites.core
  File "/usr/local/lib/python3.8/dist-packages/satellites/__init__.py", line 49, in <module>
    from .satellites_python import *
ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"
```

I already exported the path to PYTHONPATH like this in my case: 

export PYTHONPATH=/usr/local/lib/python3.8/dist-packages/

INFO: I installed gr-satellites from source code (latest state) and I installed gnu radio 3.10 from ppa

RubenMardonesVarela, 2023-07-11:
Thanks for the answer. 
I am using Ubuntu 20.04 and I am afraid that it is not available, I get "unable to locate gr-satellites" after adding the repository, that is why I tried to install it from sources. 
On the other hand, you can find the build output in the attached file: 

[BuildOutput.txt](https://github.com/daniestevez/gr-satellites/files/12011083/BuildOutput.txt)

RubenMardonesVarela, 2023-07-11:
I have searching and maybe the problem is in the pybind11 version. For my ubuntu I have: 

Version: 2.4.3-2build2

It seems I cannot upgrade it

daniestevez, 2023-07-11:
This can be indeed a problem with pybind11. In some cases we've found that the GNU Radio build in the PPA has compatibility problems with the pybind11 version in the distro. In this case, the problem would be with the PPA package. Perhaps building GNU Radio from source is a good alternative.

The gr-satellites CI for GNU Radio 3.10 is run on Ubuntu 22.04, so I can't confirm if Ubuntu 20.04 is broken.

closed 2023-07-11 by RubenMardonesVarela