daniestevez/gr-satellites #480: Can´t Decode QO-100

mdmpereira1, 2023-07-14:
Hello Folks
I am doing a project to decode the multimedia beacon of the QO-100, similar to those posted here. But when I try to run it the following error occurs:
Reed-Solomon decode fail (interleaver path 0);
Besides that the constellation that arrives is missing the symbol for bit '0':

[image]

I tried also to use the example posted here (qo100_multimedia_beacon_df2et.grc) to see if the problem was with my fluxogram, but the output is similar:

[image]

I was doing this project in GNU Radio Companion in Windows 10, but after seeing a similar issue on #217 I switched to a virtual machine  (Unbuto) and did the steps provided there, but yet again the same 2 problems appeared. 

The signal that I receive is downconverted to 432.5MHz, passes through a HackRF One to my computer. I am not sure if it is relevant but the system also has a QO-100 DX patrol. 

Thank you in advance

Miguel

daniestevez, 2023-07-15:
Hi Miguel. From a quick look, you're not correctly tuned to the signal. In the spectrum plot, the signal and its mirror image (blue and red traces) should overlap as well as possible. In that way, you can be certain that the signal is in the centre of the receiver's passband.

mdmpereira1, 2023-07-17:
Hello Daniel. Thank you for your response. I did what you suggested and is started to show the bit in the middle I think:

[image]
[image]

I added 2 frequency offsets one in the order of [-300, 300]KHz and another in the order of [-100, 100]Hz int the Fir Filter but I can't get the signal correctly, although it is close.

[image]

I think the problem is with the phase but I don´t know how to correct it.

daniestevez, 2023-07-17:
The think you now have in the spectrum plot in GNU Radio doesn't look at all like the beacon signal. See [this](https://www.florian-wolters.de/posts/qo100-hsmodem-gnuradio-decoder/) for how the spectrum should look like.

Also, it's difficult to say, since you're showing 8 MHz bandwidth in GQRX, and the beacon is only 2.7 kHz wide, but I would say that what you're seeing in GQRX is not QO-100's NB transponder, but something else.

daniestevez, 2023-09-26:
@mdmpereira1 do you have updates about this or shall we close the issue?

daniestevez, 2023-12-03:
Closing, since there have been no updates in months.