daniestevez/gr-satellites #482: Building debian bookworm package fails

kng, 2023-07-20:
Not sure if this can be fixed here or not.

I have been building debian packages for gr_satellites in the [satnogs project](https://build.opensuse.org/package/show/home:librespace:satnogs-unstable/gr-satellites) that is based on the maint-3.8 on bullseye.
However, when porting it to bookworm I stumble on this issue of dh_python not finding env DEB_HOST_ARCH.
On the [OBS](https://build.opensuse.org/package/binaries/home:knegge:branches:home:librespace:satnogs-unstable/gr-satellites_bookworm/Debian_12) it builds fine, but on a regular workstation it fails.

Running `fakeroot ./debian/rules binary` results in this in the last steps of the build:
```
   dh_python3
Traceback (most recent call last):
  File "/usr/bin/dh_python3", line 292, in <module>
    main()
  File "/usr/bin/dh_python3", line 288, in main
    dh.save()
  File "/usr/share/dh-python/dhpython/debhelper.py", line 325, in save
    self.save_autoscripts()
  File "/usr/share/dh-python/dhpython/debhelper.py", line 240, in save_autoscripts
    arch = environ['DEB_HOST_ARCH']
           ~~~~~~~^^^^^^^^^^^^^^^^^
  File "<frozen os>", line 679, in __getitem__
KeyError: 'DEB_HOST_ARCH'
make: *** [debian/rules:4: binary] Error 1
```

Running it with `DEB_HOST_ARCH=amd64 fakeroot ./debian/rules binary` solves the problem.

daniestevez, 2023-07-20:
I don't know much details about how the Debian build system works, so I'm afraid I can't be of much help here. Googling I found that [`DEB_HOST_ARCH` should be set up by a call to `dpkg-architechture`](https://linux.die.net/man/1/dpkg-architecture). Maybe this is what's missing in your build system? Perhaps Maitland Bottoms might be able to help. I don't know if he's in Github, but he's in the GNU Radio chat. He is the Debian packager for GNU Radio and other ham radio related packages.

On a more higher level note, I wasn't aware that gr-satellites was being packaged for several Debian releases (or maybe I knew but I forgot). This makes me wonder if there is something we can do to improve how this is done. Currently I'm packaging new releases of gr-satellites manually for the [gr-satellites PPA](https://launchpad.net/~daniestevez/+archive/ubuntu/gr-satellites/). The `debian/` folder to make these packages is in the repo. I upload the source packages to Launchpad, which then builds from multiple platforms. There is also a [gr-satellites package in Debian](https://packages.debian.org/sid/gr-satellites), but it is rather old. There is the [same old package in Ubuntu](https://packages.ubuntu.com/kinetic/gr-satellites). Speaking with Maitland about how to ship gr-satellites in Debian, he made a good point that the release cycle of gr-satellites is much faster than that of Debian stable, and by the point that a version arrives into Debian stable, many of the supported satellites have decayed or no longer work, while most users want the latest version to keep up with new launches. But perhaps trying to have the Debian unstable version up to date would reduce the effort needed to build for other Debian-based distributions (stable backports and Ubuntu).

I think that with good coordination we can reduce some duplicated work. At least the packages on the Opensuse build server should be listed in the gr-satellites documentation to give them more visibility, since probably few people know about them.

daniestevez, 2023-12-03:
@kng what is the status of this? Does the problem still exist?

kng, 2023-12-03:
I have installed a fresh Debian 12 and tried building both with the debian/ that comes with the package (ubuntu), and the package I use on satnogs-unstable. Both works fine, so it must be some issue with my well used VM.
Regarding the packaging etc, I am open to suggestions. That discussion can take place somewhere else thou (:

closed 2023-12-03 by kng