daniestevez/gr-satellites #511: flowgraph :warning: Could not find port pdus in:

charlez-700, 2023-08-03:
hey there.

I am learning about satellite comms and decoding, and I tried to use [the following gnuradio flowgraph for the Stratosat signal](https://www.qsl.net/k/k4kdr//grc/stratosat-fsk/). I took the information from [this tweet](https://twitter.com/scott23192/status/1680675916815425537)

I've setup the flowgraph using the provided .wav file for the stratosat signal, corrected the paths to make the files available on my system but I am getting this error message:

```
correlate_access_code_tag_bb :debug: Access code: 12ad3e
correlate_access_code_tag_bb :debug: Mask: ffffff
flowgraph :warning: Could not find port pdus in:
flowgraph :warning:   system

>>> Done (return code -11)
```

i tried to find something about "Could not find port pdus" but i had no luck at all. I know the problem is around the sync and create packed PDU module, as I disable it and the flowgraph runs, but obviously it doesn't generate any files. I am not able to debug this error further.

I am running gnuradio 3.10.5.1 and gr-satellites 5.3.0, on Ubuntu 22.04. Both packages were installed with their PPAs.

thanks for your time.

daniestevez, 2023-08-04:
I just tested that flowgraph using GNU Radio 3.10.6.0 and the current master of gr-satellites. I had to disable the UDP Source block, since that's no longer available in GNU Radio 3.10, and change the paths to the files in a couple blocks, but after this, the flowgraph worked and produced the following:
```
Qt GUI: Could not restore geometry: restoreGeometry(self, geometry: Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'NoneType'
log :warning: `pdu_to_tagged_stream` has moved to gr-pdu and will be removed from gr-blocks soon. Please update to use pdu.pdu_to_tagged_stream()
log :warning: `pdu_to_tagged_stream` has moved to gr-pdu and will be removed from gr-blocks soon. Please update to use pdu.pdu_to_tagged_stream()
vmcircbuf_sysconfig :info: Using gr::vmcircbuf_sysv_shm_factory
2023-08-04 16:36:47
Packet number 0
***** VERBOSE PDU DEBUG PRINT ******
()
pdu length =         62 bytes
pdu vector contents = 
0000: fa 53 35 b2 7f 47 7d 5d e7 b8 40 94 58 9c c1 d2 
0010: 8f 5c 40 ad ec 11 40 2f f8 40 40 6a 68 00 43 fd 
0020: 80 00 44 76 80 00 c5 e2 e1 02 82 08 01 00 80 10 
0030: 00 01 80 00 00 00 43 9a f4 e5 42 44 39 57 
************************************
2023-08-04 16:36:48
Packet number 1
***** VERBOSE PDU DEBUG PRINT ******
()
pdu length =         62 bytes
pdu vector contents = 
0000: 52 53 35 32 53 56 4d 53 57 5e 40 84 bc 6b c1 b1 
0010: d7 0a 41 53 ba 00 40 b9 8c 00 c1 a2 1c 00 44 89 
0020: 80 80 c4 05 c0 01 c7 b5 00 00 80 00 00 00 42 4e 
0030: cb c6 c2 1b c4 a0 80 00 00 00 46 c9 83 16 
************************************
``` 

I can't think of a reason why this would fail in your system, since the GNU Radio and gr-satellites versions you're using are pretty close. Maybe it helps to try disabling different parts of the flowgraph, since I'm not even sure which block is producing the `Could not find port pdus in` warning.

charlez-700, 2023-08-04:
Hi Dani,

I forgot to mention that I was running the flowgraph inside a virtual machine hosted on a M1 Macbook pro (arm arch). Did the same setup on an x86 machine and it worked as expected.

I'll close this issue for now, but if you want me to do some debugging on why this doesn't work on ARM, I would be pleased.

Thanks for your time.

closed 2023-08-04 by charlez-700

daniestevez, 2023-08-04:
Thanks! That's weird. Maybe it's not the fact that this is running in aarch64, but rather that the GNU Radio build for aarch64 is not done properly. In any case, let's keep this closed unless someone wants to do some more debugging.