daniestevez/gr-satellites #524: Mobitex-NX Telemetry forwarding (Error while submitting telemetry: HTTP Error 400: Bad Request)

janvgils, 2023-12-23:
When decoding a BEESAT-3 [observation](https://network.satnogs.org/observations/8758709/) I get a HTTP Error 400.

`gr_satellites BEESAT-3.yml --wavfile 8758709.wav --samp_rate 48e3`

The strange thing is, that the YML file doesn't have a SIDS parameter and all other observations that use the standard SatNOGS SIDS configuration don't experience this behavior.

Going through all the Mobitex related code I can't find a SIDS parameter but I get the impression that it is configured somewhere.
If this is the case, please remove it and follow the SIDS option that can be configured in the yml files.

Used YML:

```
name: BEESAT-3
norad: 39135
data:
  &tlm Telemetry:
    unknown
transmitters:
  4k8 FSK downlink:
    frequency: 435.950e+6
    modulation: FSK
    baudrate: 4800
    framing: Mobitex-NX
    data:
    - *tlm
```

Decode output:

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 1)
pdu_length = 592
contents = 
0000: 7f fd 60 41 87 cb 18 38 fa f0 42 2c 22 1a 4e 1a 
0010: e6 28 c4 9b 31 3d 14 73 3e e0 a9 57 a2 04 f3 17 
0020: b6 8b 12 39 cc f8 2d e1 b5 d5 67 00 76 27 77 36 
0030: 69 5e 9e 3d 30 82 80 99 ec b5 13 f6 47 67 2b 6e 
0040: 6f 0c 13 f0 f8 89 7b 44 72 62 4c 1c 74 60 2d fc 
0050: 63 ac 9c 55 6d 39 df 54 64 ae 2c 2c 23 e7 97 f3 
0060: 4d a4 a2 38 3b 8c f1 49 b8 6f 85 8c a4 ab 17 2d 
0070: 62 5c d1 c7 65 b5 74 9c 67 18 a8 91 4d 19 07 6a 
0080: 8b 9f 71 96 2f 7f 34 21 2c 75 8a 91 d7 55 6e 57 
0090: 35 f5 19 22 72 53 91 17 b0 b6 37 6f bf d4 ba 13 
00a0: d7 c2 05 c7 8c 13 4d b7 16 c7 81 6a ac 35 fa 8f 
00b0: 20 9b ca e0 43 da a3 28 2f 04 9f 1b 7a 9c 78 1e 
00c0: 3c 64 5c 31 e5 3e ae 21 57 90 01 91 3a 4b 0e b5 
00d0: 84 ae 13 8f 6c a8 b9 b0 5e 83 fd 46 ea ee c1 8d 
00e0: 92 a5 4a 28 a8 3c dc ec d8 b5 4a e1 b5 75 ed 72 
00f0: 5d 11 f5 e1 07 87 0b a6 aa 27 b4 d4 a4 62 73 d6 
0100: a8 47 6a 31 b0 35 d0 df ee 15 e0 d7 e6 9f a4 4c 
0110: c0 85 2d 95 38 55 78 c8 89 04 4f d8 03 86 29 74 
0120: 03 09 8f 1f 1c 40 6e c9 ac 29 63 ce b2 54 0a 68 
0130: 8f 73 76 ad b5 b9 34 1f 5c ef 78 67 14 1d b0 d7 
0140: ed 4b f7 22 1d 03 ca fa f7 4a ff 9e 81 0a e1 45 
0150: f1 67 9e 4e f8 b5 2d 2e 88 2a 51 d3 4b 7c 0c a1 
0160: f8 55 53 29 52 0e 14 04 9f e2 1d ed 5b 8d 4e 21 
0170: ca a3 72 e1 1c bd 45 14 37 02 e1 05 2c 61 e6 f0 
0180: 5c 1b b9 af f4 da d2 32 d2 7b 74 fe d1 88 e1 c2 
0190: 5e 7f 16 fa ed 76 87 25 53 c8 db 3a 23 4e 53 a4 
01a0: b5 dd 32 cd ad c0 ac da 0a ae a3 26 fa d5 9b 38 
01b0: 72 47 58 85 bd 63 d1 05 88 d2 6e a2 64 e8 f8 45 
01c0: 5d 7e a9 05 33 1e 63 4e 14 05 d7 6d b5 c1 ba 6a 
01d0: 2d 60 40 04 de 3a fd c7 6f 5e 46 8f d3 11 e3 f3 
01e0: 49 72 4e f0 b6 ec c4 e4 8b 65 14 04 3b 30 b3 d5 
01f0: a4 4b c2 7e bc 86 cd 12 e5 a2 a0 8a ce a2 8c 6b 
0200: db 14 22 85 dd 09 de de 93 8a d5 68 83 b8 20 47 
0210: b5 61 39 52 8d 1c 42 18 49 d6 2d ab 82 0b 6f ce 
0220: ae ca ab 6a 12 0b 03 f2 cc bc 46 dd 47 43 05 40 
0230: c2 d3 b9 78 5b 14 97 2f 2e ea f1 3c 71 dd 83 08 
0240: 56 6f 29 c5 26 05 e8 40 81 53 aa ff ff ff ff bb 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 1)
pdu_length = 178
contents = 
0000: 68 0d 59 de 37 0d e7 22 ed 05 e9 ac b2 de 80 84 
0010: 46 98 6c 6c e4 a4 f5 19 25 39 f9 25 df 0d 5d 38 
0020: 57 60 54 df b2 1f 2b 53 1b d4 a7 e1 c0 ff c3 81 
0030: 35 dc b4 ad d4 4d 27 30 10 67 df 69 64 42 43 ce 
0040: 3d db a7 57 fd 91 f9 e6 24 b6 c3 04 e8 2f 1a ff 
0050: ce 94 f1 e9 cd e8 b5 85 bc 04 9b 1b 99 c6 27 c1 
0060: 4e 57 1e db 0b 37 9b a1 34 ee d4 a3 9c 70 b7 49 
0070: fb b9 ab 0a 43 55 f9 ef 31 9a 8d d8 f4 b2 5e 39 
0080: 59 0b 4a 52 32 e1 98 20 3a d2 be 40 f8 ca 33 51 
0090: b4 fe cd 91 14 31 b7 de c8 ab a9 41 d5 a5 a0 c1 
00a0: 18 bf 1d 31 44 7c d2 56 50 4d 67 74 aa ff 01 00 
00b0: 00 bb 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
Error while submitting telemetry: HTTP Error 400: Bad Request
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
Error while submitting telemetry: HTTP Error 400: Bad Request
Error while submitting telemetry: HTTP Error 400: Bad Request
Error while submitting telemetry: HTTP Error 400: Bad Request

```

daniestevez, 2023-12-23:
Hi Jan, I don't think there is another SIDS telemetry submitter that gets activated when you decode BEESAT-3. I think that these HTTP 400 errors come from the SatNOGS DB server. Perhaps the reason is that the decoder is producing some frames which have a length of zero. Can you do a capture of the HTTP traffic with Wireshark? That would show more details about the problem. I don't have a GNU Radio 3.8 installation at hand to test it myself (if I recall correctly Mobitex-NX is not available for newer versions of GNU Radio).

janvgils, 2023-12-25:
Good day Daniel,

I will have look and let you know.

janvgils, 2023-12-25:
I started a tcpdump and only saw sessions being created towards the satnogs database (https, so not able to check the payload)
So this indeed seems to be related to the SatNOGS SIDS receiver.

Thanks for the support and all the best.

closed 2023-12-25 by janvgils

daniestevez, 2023-12-25:
You can have wireshark decrypt your HTTPS if you save the SSL key log and give it to wireshark. See instructions around `SSLKEYLOGFILE` in https://www.comparitech.com/net-admin/decrypt-ssl-with-wireshark/ for instance.

janvgils, 2023-12-25:
Thanks for sharing that information, what a great way to learn new technology.

It works like a charm and is confirming the frame field is empty.

```
Request URI Query: noradID=39135&source=PE0SAT&locator=longLat&longitude=x.xx&latitude=x.xx&version=1.6.6&frame=&timestamp=2023-12-25T11%3A30%3A30.080Z
    Request URI Query Parameter: noradID=39135
    Request URI Query Parameter: source=PE0SAT
    Request URI Query Parameter: locator=longLat
    Request URI Query Parameter: longitude=x.xx
    Request URI Query Parameter: latitude=x.xx
    Request URI Query Parameter: version=1.6.6
    Request URI Query Parameter: frame=
    Request URI Query Parameter: timestamp=2023-12-25T11%3A30%3A30.080Z
```
And the reply is:

```
HTTP/1.1 400 Bad Request\r\n
    [Expert Info (Chat/Sequence): HTTP/1.1 400 Bad Request\r\n]
        [HTTP/1.1 400 Bad Request\r\n]
```

daniestevez, 2023-12-25:
So it seems I was right that the problem are the empty frames generated by the Mobitex-NX decoder.

I've looked at the [SiDS v0.9 document](https://github.com/janvgils/sids/blob/main/docs/Dombrovski-SIDS-Simple-Downlink-Share-Convention.pdf) and it's not clear to me whether empty frames are permitted or not. Table 2 says that an HTTP error 400 shall be generated if any of the fields are empty. But the definition of the `frame` field in Table 1 allows for optional whitespaces. So I wonder what should happen with a `frame` field that only consists of whitespace. In any case, this is a corner case and the document is from 2015, so we could make a new version (together with the SatNOGS people) that makes it clear whether empty frames are allowed or not.