daniestevez/gr-satellites #525: Wrong vector type instantiation in Fixed Length To PDU block

v1tzl1, 2023-12-31:
Hi,

I tried to use the Fixed Length To PDU block with a custom block that produces a stream of floats as output (custom block -> fixedlen_to_pdu block). GRC is happy with the float to float connection, but when starting the flowgraph I get a  `ValueError: itemsize mismatch: custom_block0:0 using 4, fixedlen_to_pdu0:0 using 8`.

The `fixedlen_to_pdu block`  is instantiated with `satellites.fixedlen_to_pdu(gr.sizeof_float, 'frame_score', (2048+32), False)` (as generated by GRC). `gr.sizeof_float` is an `int`, but the C++ block itself expects a `gr::type::vector_type` which is an enum (and thus it can be implicitly constructed from int without any error) but the underlying enum values do not match the size of the type.

```
>>> from gnuradio import gr
>>> vt = gr.sizeof_float
>>> vt
4
>>> type(vt)
<class 'int'>
>>> gr.pdu.itemsize(vt)
8
>>> gr.pdu.type_from_pmt(gr.pdu.make_pdu_vector(vt, 0, 2))
<vector_type.complex_t: 4>
```

```
>>> from gnuradio import gr
>>> vt = gr.types.vector_type(gr.types.float_t)
>>> vt
<vector_type.float_t: 3>
>>> type(vt)
<class 'gnuradio.gr.gr_python.types.vector_type'>
>>> gr.pdu.itemsize(vt)
4
>>> gr.pdu.type_from_pmt(gr.pdu.make_pdu_vector(vt, 0, 2))
<vector_type.float_t: 3>
```

As you can see `gr.sizeof_float` correctly reports that a float is 4 bytes in GnuRadio, but enum value number 4 in `gr::type::vector_type` corresponds to `complex_t`, which in turn has a size of 8 bytes, not 4.

```
>>> gr.types.vector_type(3)
<vector_type.float_t: 3>
>>> gr.pdu.itemsize(int(3))
4
>>> gr.types.vector_type(4)
<vector_type.complex_t: 4>
>>> gr.pdu.itemsize(int(4))
8
```

I believe the correct way to instantiate fixedlen_to_pdu is thus with `gr.types.vector_type(gr.types.float_t)` instead of `gr.sizeof_float`. I will create a pull request to fix this.

v1tzl1, 2023-12-31:
For the sake of completeness, I also checked if any other templates use the `gr.sizeof_` variables. The only other block that does so is the selector block, which takes a `size_t itemsize` so the `gr_sizeof_` instantiation there is correct.

daniestevez, 2023-12-31:
Good catch! When preparing the PR, make sure to use the types from [grtypes.py](https://github.com/daniestevez/gr-satellites/blob/main/python/grtypes.py) so that the same GRC .yml serves for GR 3.10 and 3.9.

v1tzl1, 2023-12-31:
Wow you are fast. I did not see you comment in time, but I will gladly modify the PR.

satellites.grtypes does not import `int_t` and `short_t` (probably because GnuRadio 3.9 does not have them?). Should I use gr.types for these?

daniestevez, 2023-12-31:
> satellites.grtypes does not import int_t and short_t (probably because GnuRadio 3.9 does not have them?). Should I use gr.types for these?

Good question. The in-tree Tagged Stream to PDU (which is somewhat similar to Fixed Length to PDU) doesn't support `int_t` and `short_t` PDUs (see [GRC file](https://github.com/gnuradio/gnuradio/blob/main/gr-pdu/grc/pdu_tagged_stream_to_pdu.block.yml#L11)), even though [there are PMT uniform vectors of these types](https://github.com/gnuradio/gnuradio/blob/86af478a88cf0521b023ad2c924c74ea38d9adcf/gnuradio-runtime/include/pmt/pmt.h#L484). I don't know if there's a reason for this. It might even happen that the [`pmt::make_pdu_vector()` call](https://github.com/daniestevez/gr-satellites/blob/7a989f915ac82528c91a2afa29b4f4585bdcd881/lib/fixedlen_to_pdu_impl.cc#L106) doesn't work for these types. So I'd stay away of having Fixed Length To PDU support `int_t` and `short_t` unless there is a specific need for them, because that allows us to have the same GRC file for 3.10 and 3.9.

v1tzl1, 2024-01-01:
Thank you for your comments. I was hesitant to remove existing options at first. But given the fact that they never worked in the first place, I agree that this will not break anything that was not already broken before.

I updated the PR accordingly. The GRC file now only supports `float`, `gr_complex` and `byte` and uses `satellites.grtypes` for these.

PR #526 (2024-01-03): Fix GRC template to use correct type for fixedlen_to_pdu instantiation
fixes #525
commit 4b1b9e21 (2024-01-03): Fix GRC template to use correct type for fixedlen_to_pdu instantiation

Use satellites.grtypes to increase compatibility between GnuRadio 3.9 and 3.10.
Remove int and short options from the template, because they are not supported
by GnuRadio 3.9 and since the existing implementation was broken, it likely was
not used anywhere.

fixes #525
commit 6211b65f (2024-01-03): Fix GRC template to use correct type for fixedlen_to_pdu instantiation

Use satellites.grtypes to increase compatibility between GnuRadio 3.9 and 3.10.
Remove int and short options from the template, because they are not supported
by GnuRadio 3.9 and since the existing implementation was broken, it likely was
not used anywhere.

fixes #525

(cherry picked from commit 4b1b9e21ec9cdbc389e38cbe0e3c1a2ccfe2d91c)
commit e2886515 (2024-01-03): Fix GRC template to use correct type for fixedlen_to_pdu instantiation

Use satellites.grtypes to increase compatibility between GnuRadio 3.9 and 3.10.
Remove int and short options from the template, because they are not supported
by GnuRadio 3.9 and since the existing implementation was broken, it likely was
not used anywhere.

fixes #525

(cherry picked from commit 4b1b9e21ec9cdbc389e38cbe0e3c1a2ccfe2d91c)
closed 2024-01-03 by daniestevez