daniestevez/gr-satellites #536: MRC-100

ik8ozv, 2024-01-26:
Please anybody can help me on how to decode the MRC-100 Satellite?
If i try to run in the terminal the command:

gr_satellites mrc-100 --udp --samp_rate 48e3

i get the following error:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 322, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 307, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 196, in __init__
    self.flowgraph = satellites.core.gr_satellites_flowgraph(
  File "/usr/local/lib/python3.10/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 147, in __init__
    self._init_additional_datasinks()
  File "/usr/local/lib/python3.10/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 222, in _init_additional_datasinks
    self.get_telemetry_submitters(self.satyaml, self.config,
  File "/usr/local/lib/python3.10/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 349, in get_telemetry_submitters
    submitters.append(datasinks.telemetry_submit(
  File "/usr/local/lib/python3.10/dist-packages/satellites/components/datasinks/telemetry_submit.py", line 69, in __init__
    self.submit = bme_ws_submitter()
  File "/usr/local/lib/python3.10/dist-packages/satellites/bme_ws_submitter.py", line 28, in __init__
    self.ws = websocket.WebSocket()
TypeError: WebSocket.__init__() missing 3 required positional arguments: 'environ', 'socket', and 'rfile'
Exception ignored in: <function bme_ws_submitter.__del__ at 0x7f8f869eb640>
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/satellites/bme_ws_submitter.py", line 41, in __del__
    if self.ws is not None:
  File "/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py", line 133, in __getattr__
    return getattr(self.gateway, name)
AttributeError: 'gnuradio.gr.gr_python.block_gateway' object has no attribute 'ws'
Segmentation fault (core dumped)

Please where am i missing?
How ro fix the problem
Thanks
Enzo

daniestevez, 2024-01-26:
What version of gr-satellites are you using, in which system, and how have you installed it?

Also include the output of `pip list`.

ik8ozv, 2024-01-26:
Thanks Dani for reply.
I am using UBUNTU 22.04.3.
GR-Satellites should be the latest one. Please can you tell me how to che the version and also how to 
get the output of pip list to include.
Thanks a lot
Enzo

ik8ozv, 2024-01-28:
@daniestevez
Dani please any news for me?
Thanks
Enzo

daniestevez, 2024-01-29:
> websocket 0.2.1

I think that the problem is that you have the wrong websocket package installed in your system. This is what I have:

```
websocket-client          1.6.4
websockets                10.4
```

The package required for this is [websocket-client](https://pypi.org/project/websocket-client/) rather than [websocket](https://pypi.org/project/websocket/).

ik8ozv, 2024-01-29:
Hello Dani,
Thanks for your help it worked.
Now when i run:

gr_satellites Mrc-100 --udp --samp_rate 48e3
i get:

enzo@enzo-virtual-machine:~$ gr_satellites Mrc-100 --udp --samp_rate 48e3
correlate_access_code_tag_ff :debug: Access code: e31c9dae
correlate_access_code_tag_ff :debug: Mask: ffffffff
correlate_access_code_tag_ff :debug: Access code: e31c9dae
correlate_access_code_tag_ff :debug: Mask: ffffffff
correlate_access_code_tag_ff :debug: Access code: e31c9dae
correlate_access_code_tag_ff :debug: Mask: ffffffff
correlate_access_code_tag_ff :debug: Access code: e31c9dae
correlate_access_code_tag_ff :debug: Mask: ffffffff
udp_source :info: Listening for data on UDP port 7355.

But i do not see anything on the screen when receiving the satellite signal 
and when there is not signal from the satellite i see:

correlate_access_code_tag_ff :debug: writing tag at sample 5193
correlate_access_code_tag_ff :debug: writing tag at sample 35974
correlate_access_code_tag_ff :debug: writing tag at sample 172324
correlate_access_code_tag_ff :debug: writing tag at sample 43918
correlate_access_code_tag_ff :debug: writing tag at sample 278872

This happens also on other satellites.
Please any hint?
Thanks
73's de Enzo IK8OZV

daniestevez, 2024-01-29:
Probably the signal that gets passed to gr-satellites from GQRX is not good enough, as it has been the case in other issues you've opened.

daniestevez, 2024-01-29:
The false syncword detections (`correlate_access_code_tag_ff :debug:` messages) are to be expected.

ik8ozv, 2024-01-29:
But i can decode with the same signal on Windows why not on Ubuntu?

daniestevez, 2024-04-12:
Closing, since it has been months without activity and I think there are no outstanding issues. Feel free to reopen.

closed 2024-04-12 by daniestevez