daniestevez/gr-satellites #537: Why i cannot decode live??'

ik8ozv, 2024-01-31:
Hello all,
this morning I have made a try and would like to share it with you all to see if anybody
can help to solve the issue.
This is what I have done on an IRIS-A pass:

I ran in the terminal the command:
gr_satellites IRIS-A --udp --samp_rate 48e3
 and I got as response:

udp_source :info: Listening for data on UDP port 7355.

I have opened GQRX and enabled the UDP button after been sure that the network port was set as 7355.
After a couple of burst I did not get anything on the terminal so I have, enabled
the REC button and continued the pass.
After the pass I have loaded in gnuradio, the SATELLITE_DECODER.grc from the folder:

/gr-satellites/examples/satellite_decoder

I have then changed the Satellite Decoder block changing the Satellite name to IRIS-A
and the Wav file source block pointing it the recorded file.
Ran it but did not see anything so I have tried to switch from NO to YES the Repeat field in the Wav File Source block.
Ran again and this time after a couple of seconds I got:

message_debug :info: The print_pdu port is deprecated and will be removed; forwarding to print.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 30 bytes
pdu vector contents =
0000: df b6 f7 1f 67 0c 5d 03 1f 80 62 89 5c 33 1d 05
0010: 96 b6 29 84 e3 30 3d fb bf a0 19 f2 4c e0

message_debug :info: The print_pdu port is deprecated and will be removed; forwarding to print.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 32 bytes
pdu vector contents =
0000: 00 23 6d cd 4a b7 9c 9e 13 12 8e 1a b8 75 94 67
0010: a3 f6 2f 68 89 49 11 2d 79 77 6f b9 33 66 a4 d8

message_debug :info: The print_pdu port is deprecated and will be removed; forwarding to print.
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 16 bytes
pdu vector contents =
0000: a0 46 65 f5 3c c6 6c 9e b7 16 25 d5 ca b6 50 ad

Please why I do not get anything when I run in the terminal:

gr_satellites IRIS-A --udp --samp_rate 48e3

Thanks a lot
73's de Enzo IK8OZV

daniestevez, 2024-01-31:
As I've indicated in other of your issues, probably the signal quality that you're feeding into `gr_satellites` is not good enough to produce decodes, or you have GQRX misconfigured (as in using the wrong demodulation mode or not correctly tuned in frequency). My advice if you want to troubleshoot this is that you record data so that you can replicate (either record a WAV file with GQRX or record the UDP packets with wireshark), and then run `gr_satellites` on this data while dumping the internal signals as [indicated in the documentation](https://gr-satellites.readthedocs.io/en/latest/command_line.html#dump-internal-signals) and produce some plots to check the signal quality.

ik8ozv, 2024-01-31:
Dani,
you keeping saying that "probably the signal quality that you're feeding into gr_satellites is not good enough to produce decodes"
Right now i have done another test.
I had a pass of GOMX-1. I have recorded just a couple of burst and saved them while the satellite was still going on.
I ran them in the satellite_decoder.grc and got the incoming data. 
The only warning i get is:

This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

But i did not understood where to place it and how to connect it as i get always red lines.

I closed the terminal program and while the satellite was still sending data i ran the GOMX-1.grc flowgraph and i do not get
any incoming data. I think it is not a signal problem.
Please any hint?
Thanks
73's de Enzo IK8OZV

daniestevez, 2024-02-01:
Can you share the data you recorded?

ik8ozv, 2024-02-01:
Sure i can.
I have saved 4 files:

INS-2TD
CHOMPTT
GRBALPHA
LUME-1

If you tell me how to do to share them, i will.
Thanks

daniestevez, 2024-02-01:
I guess you can upload them somewhere and share a link.

ik8ozv, 2024-02-01:
Dani i have loaded a wav.rar file on https://files.fm/.
Here is the link:

https://files.fm/u/e44g45mw4n

Please let me know if you can download the file.
Wait for your reply
Thanks
Enzo

ik8ozv, 2024-02-02:
But I still did not understand why I do not get any data when I run from the flowgraph. Maybe how you said the audio in the GQRX is to high. How has to be set? How to know which is the correct position of the volume please?
Thanks

daniestevez, 2024-02-06:
> How has to be set? How to know which is the correct position of the volume please?

Adjust it to avoid clipping the output. I was looking at your WAVs in Audacity, but maybe there's a quicker method.

daniestevez, 2024-04-12:
Closing, since the problem was diagnosed as input files being clipped and there hasn't been any activity in months. Feel free to reopen.

closed 2024-04-12 by daniestevez