daniestevez/gr-satellites #539: Depraction of distutils

winnieXY, 2024-02-06:
distutils will be removed with python 3.12 - please switch to shutils.which instead. See below for a short patch..

With best regards
Patrick

[clang_format.txt](https://github.com/daniestevez/gr-satellites/files/14185476/clang_format.txt)

daniestevez, 2024-02-06:
Thanks for sending this.

It's probably worth to pull the latest version of https://github.com/gnuradio/gnuradio/blob/main/tools/clang_format.py, since it has a few other changes besides replacing `distutils` by `shutil`. Though the script might as well be removed altogether. It is not used in CI and I don't know if anyone actually uses it.

On a related note, it would be good to update the [clang format CI workflow](https://github.com/daniestevez/gr-satellites/blob/main/.github/workflows/clang-format-lint.yml) to use the latest version of https://github.com/DoozyX/clang-format-lint-action.

Could you send a PR with these changes?

daniestevez, 2024-11-02:
Marking as closed. The script has been removed in ac24de06fe43fa2e36ef772215b3da2a3bbbb27f. The clang-format-lint-action is not used any more since cbe4dc1c736f13984ac49989a0d42917bdab8cd1.

closed 2024-11-02 by daniestevez