daniestevez/gr-satellites #540: Error Runnig Flowgraphs

ik8ozv, 2024-02-11:
Hello,
i am trying to run Folwgraphs on Ubuntu 13.10.1 and i am noticing this 2 errors:

QSocketNotifier: Can only be used with threads started with QThread

and 
gr::vmcircbuf :error: shmat (3): Argomento non valido

Please anybody can tell me how to solve?
Thanks
73's de Enzo IK8OZV

daniestevez, 2024-02-12:
I guess you mean Ubuntu 23.10.1.

These errors look like warnings rather than fatal errors. Are they fatal errors? Also, they seem to be caused by GNU Radio, not gr-satellites specific. Do they also happen if you try to run a simple GNU Radio flowgraph that does not involve gr-satellites?

daniestevez, 2024-04-12:
Closing, as it needs more information not provided by @ik8ozv and it doesn't seem to be a issue specific to gr-satellites.

closed 2024-04-12 by daniestevez