daniestevez/gr-satellites #543: Unable to compile for GNU Radio 3.8 on Ubuntu 22.04

kongschlong, 2024-04-11:
Hello,

I am trying to set up a ground station that depends on GNU Radio 3.8 (some custom OOT modules have not been ported yet).

Since I'd like to avoid being stuck on Ubuntu 20.04, I am compiling GNU Radio 3.8 from source on 22.04. This is all running in a docker container on an `amd64` host. I will include relevant parts of the Dockerfile below. From what I can tell, the program is being built successfully.

Unfortunately I am unable to compile gr-satellites on 22.04 with my self-compiled GNU Radio, it worked just fine on 20.04 with the GNU Radio version from the repository. It appears that the compiler is unable to find Volk?

```
[ 61%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/manchester_sync_impl.cc.o
8.831 In file included from /home/groundstation/oot-modules/gr-satellites/lib/manchester_sync_impl.cc:10:
8.831 /home/groundstation/oot-modules/gr-satellites/lib/manchester_sync_impl.h:14:10: fatal error: volk/volk_alloc.hh: No such file or directory
8.831    14 | #include <volk/volk_alloc.hh>
8.831       |          ^~~~~~~~~~~~~~~~~~~~
8.831 compilation terminated.
```

The VolkConfig.cmake files seem to be in the correct location:

```
root@826ea153cfd1:/usr/local/lib/cmake/volk# ls
VolkConfig.cmake  VolkConfigVersion.cmake  VolkTargets-release.cmake  VolkTargets.cmake
```

I can also see `libvolk.so` and `libvolk.so.2.0` in `/usr/local/lib`.

I have to admit that I'm not very familiar with C, I don't really understand how the build scripts work, so I'm thankful for any help.

---

As promised, here is my Dockerfile:

```Dockerfile
FROM ubuntu:22.04

# get all dependencies
RUN apt update && DEBIAN_FRONTEND=noninteractive apt install -y \
    # dependencies for building gnu-radio from source
    git cmake g++ libboost-all-dev libgmp-dev swig python3-numpy \
    python3-mako python3-sphinx python3-lxml doxygen libfftw3-dev \
    libsdl1.2-dev libgsl-dev libqwt-qt5-dev libqt5opengl5-dev python3-pyqt5 \
    liblog4cpp5-dev libzmq3-dev python3-yaml python3-click python3-click-plugins \
    python3-zmq python3-scipy python3-gi python3-gi-cairo gir1.2-gtk-3.0 \
    libcodec2-dev libgsm1-dev libusb-1.0-0 libusb-1.0-0-dev libudev-dev \
    # not sure what this is for
    gir1.2-gtk-3.0 \
    # for the oot modules
    libboost-all-dev libcppunit-dev liblog4cpp5-dev swig liborc-dev libgsl-dev python3-pip \
    [...]

# -----  Build gnu-radio from source  ----------------------------------------#
RUN mkdir -p /tim-setup/gnuradio
WORKDIR /tim-setup/gnuradio

RUN git clone https://github.com/gnuradio/gnuradio.git
WORKDIR gnuradio
RUN git checkout maint-3.8 && git submodule update --init --recursive && mkdir build
WORKDIR build
RUN cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../
RUN make -j$(nproc)
RUN make install && ldconfig

ENV PYTHONPATH=/usr/local/lib/python3/dist-packages:/usr/local/lib/python3/site-packages:$PYTHONPATH
ENV LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
RUN ldconfig

# -----  install flowgraph dependencies  ----------------------------------#

# install additionally required python packages for gr-satellites
RUN pip3 install construct requests websocket-client
ENV PYTHONPATH "${PYTHONPATH}:/usr/local/lib/python3/dist-packages"

WORKDIR /home/groundstation

# acquire the required OOT modules
COPY oot-modules/ oot-modules

# compile and install the OOT modules
WORKDIR oot-modules
[...]

WORKDIR gr-satellites
RUN mkdir build
WORKDIR build
RUN cmake -DVolk_Dir=/usr/local/lib/cmake/volk ..
RUN make
RUN make install
RUN ldconfig

[...]
```

daniestevez, 2024-04-12:
Have you checked that `volk/volk_alloc.hh` is installed somewhere (maybe in `/usr/local/include`)? If that's the case, then it looks like `gcc` isn't getting the right `-I` flag for the include path (say `-I/usr/local/include`), though I don't know why cmake isn't setting that up correctly, given that it doesn't complain that it hasn't found Volk.

kongschlong, 2024-04-12:
Hi, thanks for your reply!

The folder `/usr/local/include/volk/` exists, but interestingly `volk_alloc.hh` is missing. Does this mean there's an issue with my GR build?

```
groundstation@2d98d8e69f83:~$ ls /usr/local/include/volk/
asm                                            volk_32f_s32f_multiply_32f.h                    volk_32fc_x2_dot_prod_32fc.h
constants.h                                    volk_32f_s32f_normalize.h                       volk_32fc_x2_multiply_32fc.h
saturation_arithmetic.h                        volk_32f_s32f_power_32f.h                       volk_32fc_x2_multiply_conjugate_32fc.h
volk.h                                         volk_32f_s32f_s32f_mod_range_32f.h              volk_32fc_x2_s32f_square_dist_scalar_mult_32f.h
volk_16i_32fc_dot_prod_32fc.h                  volk_32f_s32f_stddev_32f.h                      volk_32fc_x2_square_dist_32f.h
volk_16i_branch_4_state_8.h                    volk_32f_sin_32f.h                              volk_32i_s32f_convert_32f.h
volk_16i_convert_8i.h                          volk_32f_sqrt_32f.h                             volk_32i_x2_and_32i.h
volk_16i_max_star_16i.h                        volk_32f_stddev_and_mean_32f_x2.h               volk_32i_x2_or_32i.h
volk_16i_max_star_horizontal_16i.h             volk_32f_tan_32f.h                              volk_32u_byteswap.h
volk_16i_permute_and_scalar_add.h              volk_32f_tanh_32f.h                             volk_32u_byteswappuppet_32u.h
volk_16i_s32f_convert_32f.h                    volk_32f_x2_add_32f.h                           volk_32u_popcnt.h
volk_16i_x4_quad_max_star_16i.h                volk_32f_x2_divide_32f.h                        volk_32u_popcntpuppet_32u.h
volk_16i_x5_add_quad_16i_x4.h                  volk_32f_x2_dot_prod_16i.h                      volk_32u_reverse_32u.h
volk_16ic_convert_32fc.h                       volk_32f_x2_dot_prod_32f.h                      volk_64f_convert_32f.h
volk_16ic_deinterleave_16i_x2.h                volk_32f_x2_fm_detectpuppet_32f.h               volk_64f_x2_add_64f.h
volk_16ic_deinterleave_real_16i.h              volk_32f_x2_interleave_32fc.h                   volk_64f_x2_max_64f.h
volk_16ic_deinterleave_real_8i.h               volk_32f_x2_max_32f.h                           volk_64f_x2_min_64f.h
volk_16ic_magnitude_16i.h                      volk_32f_x2_min_32f.h                           volk_64f_x2_multiply_64f.h
volk_16ic_s32f_deinterleave_32f_x2.h           volk_32f_x2_multiply_32f.h                      volk_64u_byteswap.h
volk_16ic_s32f_deinterleave_real_32f.h         volk_32f_x2_pow_32f.h                           volk_64u_byteswappuppet_64u.h
volk_16ic_s32f_magnitude_32f.h                 volk_32f_x2_s32f_interleave_16ic.h              volk_64u_popcnt.h
volk_16ic_x2_dot_prod_16ic.h                   volk_32f_x2_subtract_32f.h                      volk_64u_popcntpuppet_64u.h
volk_16ic_x2_multiply_16ic.h                   volk_32f_x3_sum_of_poly_32f.h                   volk_8i_convert_16i.h
volk_16u_byteswap.h                            volk_32fc_32f_add_32fc.h                        volk_8i_s32f_convert_32f.h
volk_16u_byteswappuppet_16u.h                  volk_32fc_32f_dot_prod_32fc.h                   volk_8ic_deinterleave_16i_x2.h
volk_32f_64f_add_64f.h                         volk_32fc_32f_multiply_32fc.h                   volk_8ic_deinterleave_real_16i.h
volk_32f_64f_multiply_64f.h                    volk_32fc_conjugate_32fc.h                      volk_8ic_deinterleave_real_8i.h
volk_32f_8u_polarbutterfly_32f.h               volk_32fc_convert_16ic.h                        volk_8ic_s32f_deinterleave_32f_x2.h
volk_32f_8u_polarbutterflypuppet_32f.h         volk_32fc_deinterleave_32f_x2.h                 volk_8ic_s32f_deinterleave_real_32f.h
volk_32f_accumulator_s32f.h                    volk_32fc_deinterleave_64f_x2.h                 volk_8ic_x2_multiply_conjugate_16ic.h
volk_32f_acos_32f.h                            volk_32fc_deinterleave_imag_32f.h               volk_8ic_x2_s32f_multiply_conjugate_32fc.h
volk_32f_asin_32f.h                            volk_32fc_deinterleave_real_32f.h               volk_8u_conv_k7_r2puppet_8u.h
volk_32f_atan_32f.h                            volk_32fc_deinterleave_real_64f.h               volk_8u_x2_encodeframepolar_8u.h
volk_32f_binary_slicer_32i.h                   volk_32fc_index_max_16u.h                       volk_8u_x3_encodepolar_8u_x2.h
volk_32f_binary_slicer_8i.h                    volk_32fc_index_max_32u.h                       volk_8u_x3_encodepolarpuppet_8u.h
volk_32f_convert_64f.h                         volk_32fc_magnitude_32f.h                       volk_8u_x4_conv_k7_r2_8u.h
volk_32f_cos_32f.h                             volk_32fc_magnitude_squared_32f.h               volk_avx2_intrinsics.h
volk_32f_expfast_32f.h                         volk_32fc_s32f_atan2_32f.h                      volk_avx_intrinsics.h
volk_32f_index_max_16u.h                       volk_32fc_s32f_deinterleave_real_16i.h          volk_common.h
volk_32f_index_max_32u.h                       volk_32fc_s32f_magnitude_16i.h                  volk_complex.h
volk_32f_invsqrt_32f.h                         volk_32fc_s32f_power_32fc.h                     volk_config_fixed.h
volk_32f_log2_32f.h                            volk_32fc_s32f_power_spectrum_32f.h             volk_cpu.h
volk_32f_null_32f.h                            volk_32fc_s32f_x2_power_spectral_density_32f.h  volk_malloc.h
[... cut ...]
volk_32f_s32f_calc_spectral_noise_floor_32f.h  volk_32fc_s32fc_rotatorpuppet_32fc.h            volk_prefs.h
volk_32f_s32f_convert_16i.h                    volk_32fc_s32fc_x2_rotator_32fc.h               volk_sse3_intrinsics.h
volk_32f_s32f_convert_32i.h                    volk_32fc_x2_add_32fc.h                         volk_sse_intrinsics.h
volk_32f_s32f_convert_8i.h                     volk_32fc_x2_conjugate_dot_prod_32fc.h          volk_typedefs.h
volk_32f_s32f_mod_rangepuppet_32f.h            volk_32fc_x2_divide_32fc.h
```

daniestevez, 2024-04-12:
Apparently [`volk_alloc.hh` first appeared in Volk 2.1.0](https://github.com/gnuradio/volk/commit/4cc47ea71b4c7d3f9433156f5f61b6c7e11b4f53). In the `main-3.8` branch of GNU Radio, the `volk` submodule points to Volk 2.0.0, so that's probably the version of Volk that you've built and installed. On the other hand, Ubuntu 20.04 [ships with Volk 2.2.1](https://packages.ubuntu.com/focal/libvolk2.2), so that's the reason why the CI of the `maint-3.8` branch of gr-satellites, which [builds against Ubuntu 20.04's GNU Radio and Volk](https://github.com/daniestevez/gr-satellites-ci-action/blob/maint-3.8/Dockerfile) isn't hitting this problem.

kongschlong, 2024-04-12:
That's interesting. So what's the best way to proceed now? Can I just manually move the `volk` submodule to 2.2.1 before compiling? Or do you have any other suggestions for running on 22.04? (The only reason I'm compiling from source is because I cannot find a PPA for GR 3.8, I don't need the flexibility of compiling myself)

daniestevez, 2024-04-12:
My advice would really be to move away from GNU Radio 3.8 and use GNU Radio 3.10 instead, because GR 3.8 is unmaintained and outdated. Whatever OOT modules that you're using that depend on 3.8 should be ported to 3.10. Otherwise they're just technical debt.

But if you really want to stick with GR 3.8, then pointing the `volk` submodule to whatever version you want to use seems a valid approach. It's been a long time I don't use GR 3.8, but I can't think of a reason why it shouldn't work with even the latest stable release of Volk.

kongschlong, 2024-04-12:
I fully agree, I'd also like to move on from 3.8. Unfortunately it's not my project and I have no experience with the development of OOT modules. I'm just tasked with making it work.

For now, I've updated `volk` to 2.1 (I was having some issues with a later version) and it's compiling now. Thanks for your help!

---

For reference:

```bash
cd volk
git fetch --tags && git checkout v2.1.0
cd ..
```

closed 2024-04-12 by kongschlong