daniestevez/gr-satellites #549: CCSDS Concatenated Deframer does not run on gnuradio-companion 3.10.7.0

F4HVO, 2024-06-14:
Hello, I am trying to run the example ccsds_deframer.grc on gnuradio 3.10.7.0. My host is a docker image of ubuntu 22.04.

The sketch does not start and I have the error :

```
> python3 ccsds_deframer.py
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-gnuradio'
flowgraph :warning: Could not find port pdus in:
flowgraph :warning:   system
Segmentation fault
```

I did not find any occurence of 'pdus' in the python script nor in the xml/python files of the ccsds_concatenated_deframer block.

This happens only when I use the ccsds_concatenated_deframer block.

daniestevez, 2024-06-14:
I have tested this in my system and it doesn't give that warning and doesn't segfault. However I noticed that the example wasn't decoding any packets. The reason is that the Reed Solomon basis was set incorrectly, probably because of some changes to the GRC block that have happened since the flowgraph was made. I've fixed this in 20d1841cd3dec51551db3066fb332d240ebf9af4.

Regarding your problem, how have you installed GNU Radio and gr-satellites? If you've installed both from their respective PPAs, then that's the problem. The gr-satellites PPA is built against the GNU Radio in Ubuntu, not in the GNU Radio PPA. If you mix both PPAs you'll get ABI incompatibilities.

F4HVO, 2024-06-14:
Yes the problem originated from my installation. I was installing GNU Radio and gr-satellites from their respective PPAs.

I switched to gr-satellites installation from source (pulling master to try your latest commit) and it seems to work perfectly: the example sketch runs without error and decodes data.
Thank you very much. Problem solved !

closed 2024-06-14 by F4HVO