daniestevez/gr-satellites #587: Setting ax.25 deframer G3RUH to False

N6RFM, 2024-07-11:
Hi Dani,

The new satellite Robusta-3a decodes using the the recently updated satyaml file.  Thank you.

However, when examined using Direwolf the source and dest information seem corrupted.
dest    `.F4KJ  8 c/r=1 res=1 last=0
 source   pFX6F  2 c/r=1 res=1 last=0

Using a legacy flowgraph with a fsk decoder block and the  ax.25 deframer block G3RUH set to False five setting G3RUH  gives
dest    F4KJX   0 c/r=1 res=3 last=0
 source  FX6FRD  0 c/r=1 res=3 last=1

So, within the satyaml file is there a way to actively set G3RUH scrambling to False?

Thanks as always!
Bob
N6RFM

daniestevez, 2024-07-11:
The way to disable G3RUH in SatYAML is simply to specify `framing: AX.25` instead of `framing: AX.25 G3RUH`, which is what the SatYAML for ROBUSTA-3A does.

Your problem does not seem related to G3RUH. It looks like the packets are right but have 1 or 2 extra characters at the beginning. Maybe Direwolf is interpreting wrongly the timestamps that gr-satellites includes in the KISS stream. Does the telemetry parser in gr-satellite print out correct callsigns? It did for me when I tested.

N6RFM, 2024-07-14:
Thanks Dani!
Works FB now.  Seems some of the frames I was trying to decode had marginal SNR.
Cheers,
Bob

closed 2024-07-14 by N6RFM