daniestevez/gr-satellites #634: gr-satellites Debian build gnuradio 3.10.11.0 ModuleNotFoundError: No module named satellites.bindings

janvgils, 2024-08-05:
As the title describes, when building gr-satellites on Debian I run into a no module named satellites.bindings and the build fails.

In the past we had something similar that was related to pybind11-dev but looking at the version history and gnuradio-config this doesn't seem to be the case with this build fail.

All information can be found at the following location.

daniestevez, 2024-08-05:
I've seen something like this when the pybind11 that was packaged was not the same that was used to build the GNU Radio package. It seems that something in this sense is broken in your Debian testing. Are you able to build any other GNU Radio OOT modules that contain C++ block?

janvgils, 2024-08-05:
Indeed we ran into this before with GNURadio 3.9 some years ago, that is why I added the gnuradio-config-info information.

`gnuradio-config-info --version --pybind`

```
3.10.11.0
2.12.0
```

The pybind11-dev package is also 2.12 `pybind11-dev 2.12.0-1` but maybe there is a slight sub version difference.
I will keep an eye on possible updates in the next couple of days and if that takes to long try to build my own.

It would be nice if it was possible to view all the package version that where used when gnuradio was build.

janvgils, 2024-08-09:
The last couple of days I haven't seen any updates regarding pybind11 nor gnuradio on Debian testing/trixie.

So in the meantime, I am trying to create some debug logging that would help me in creating a Debian bug report.

Both `cmake -DCMAKE_BUILD_TYPE=Debug ../` nor `make -d` have any information that could help in creating the report.

Can anybody point me in a direction that would show extra information in why `satellites.bindings` is failing?

Log information can be found at this [location](https://www.pe0sat.vgnet.nl/download/gr-satellites/issue-634/)

janvgils, 2024-08-11:
After this evenings updates, gr-satellites is building and working again:

```
Upgrading:                      
  bash                  libgirepository-2.0-0  libglib2.0-data
  busybox               libglib2.0-0t64        libglib2.0-dev
  gir1.2-glib-2.0       libglib2.0-0t64:i386   libglib2.0-dev-bin
  libglib2.0-bin

```

I guess one of the glib packages solved the issue.

closed 2024-08-11 by janvgils

daniestevez, 2025-07-22:
This is definitely an Arch Linux bug. I'm glad I've not updated any of my Arch systems yet and being hit by this bug, as it would force me to revert pybind11. Definitely a major release update of pybind11 requires a rebuild of all packages. GNU Radio is even more picky because since people will probably build OOTs against system pybind11, system pybind11 needs to be both ABI and API compatible with GNU Radio.