daniestevez/gr-satellites #659: Unify cc11xx_packet_crop and sx12xx_packet_crop

daniestevez, 2024-09-05:
These two blocks do nearly the same thing. They should be marked as deprecated and a new block that does the same as sx12xx_packet_crop and is called something like length_field_packet_crop should be added to replace them.

mattcattb, 2024-09-06:
This looks very interesting. I would love to work on this!

daniestevez, 2024-09-06:
Go for it. A pull request is welcome.

mattcattb, 2024-09-10:
Hi stevez. I would love to contribute, but first I need some help with installation and setup. Which method of installation is perfered, from source, from conda, or from the ubuntu PPA? I am using ubuntu jelly.

mattcattb, 2024-09-18:
Hi, I have been looking into the 2 libraries and it seems I would only need to update binar2_deframer, fossasat_deframer, grizu263a_deframer for sx12xx and then aaltol_deframer, binarl_deframer, endurosat_deframer, reaktor_hello_world_deframer for cc11xx? I would also need to update the init and yaml files for this new length_field_packet_crop?

daniestevez, 2024-09-18:
Yes. A GRC YAML file needs to be added and the Python `__init__.py` files and `CMakeLists.txt`  need to be updated as for any other GNU Radio block that is added.

mattcattb, 2024-09-18:
I believe I have just finished this feature, I am just first trying to test out the normal main build. I am running into an error, after `mkdir build`

daniestevez, 2024-09-19:
Since that is working on CI, it suggests an issue with your system building Python bindings. What distribution are you using, what version of GNU Radio and how did you install it, etc.? You can do some investigation in `build/python/bindings` to see if `satellites_python*.so` has been built, is correctly linked (`ldd`), etc. Also watch out for anything that could have looked like an error or warning about Python bindings during the build process.

mattcattb, 2024-09-19:
The following is my list of steps:
1. create conda enviroment: conda env create -n gr-sat python=3.9, then activate that
2. git clone main branch of repository
3. pip3 install --user --upgrade construct requests
4. pip install websocket-client
5. go to the cloned github 
6. `mkdir build 
cd build
cmake ..
make
sudo make install
sudo ldconfig`

daniestevez, 2024-09-20:
I'm not sure if trying to build out-of-tree modules against GNU Radio installed through conda is a good idea. I imagine that versions mismatches between the gcc or pybind11 that is installed in your distribution versus the ones that were used to build the GNU Radio conda package could cause Python bindings problems like  the ones you're seeing. Maybe someone that is more familiar with conda would give better advice, but I would recommend to install GNU Radio through your distribution package manager if you're going to build out-of-tree modules.