daniestevez/gr-satellites #671: AttributeError: module 'satellites' has no attribute 'nrzi_decode'

jane0828, 2024-09-28:
Hi, I'm working with CUAVA-2.

I installed gnuradio 3.10.7 version and got gr-satellites 5.6.0 source code from your git website.

By the way, we're keep struggling with the following issue;

-------------------------------------------------------------------------------
Generating: '/home/miman/Downloads/transceiver.py'

Executing: /usr/bin/python3 -u /home/miman/Downloads/transceiver.py

Qt GUI: Could not restore geometry: restoreGeometry(self, Union[QByteArray, bytes, bytearray]): argument 1 has unexpected type 'NoneType'
Traceback (most recent call last):
  File "/home/miman/Downloads/transceiver.py", line 422, in <module>
    main()
  File "/home/miman/Downloads/transceiver.py", line 400, in main
    tb = top_block_cls(centre_freq_rx=options.centre_freq_rx, cmd_port=options.cmd_port, decimation_rx=options.decimation_rx, ftp_port=options.ftp_port, host=options.host, lo_offset_rx=options.lo_offset_rx, local_ftp_port=options.local_ftp_port, port=options.port)
  File "/home/miman/Downloads/transceiver.py", line 105, in __init__
    self.satellites_nrzi_decode_1 = satellites.nrzi_decode()
AttributeError: module 'satellites' has no attribute 'nrzi_decode'

>>> Done (return code 1)

-------------------------------------------------------------------------------

[image]

We are using bladeRF as our SDR device.
Why is QT GUI block & NRZI decode block not working?

daniestevez, 2024-09-28:
Hi, regarding the QT GUI block, the error you get looks non fatal, so probably you can ignore it. It looks somewhat familiar, so probably I've seen and ignored it too. The correct place to report about this QT GUI bug would be in https://github.com/gnuradio/gnuradio/issues, since that's an in-tree block. It might even happen that the bug is fixed in newer versions of GNU Radio.

Regarding the `AttributeError: module 'satellites' has no attribute 'nrzi_decode'`, this points to a problem with building the Python bindings for gr-satellites. How did you install GNU Radio? What Linux distribution are you using?

jane0828, 2024-09-28:
Hi, thank you so much for fast reply.

I installed GNUradio 3.10.7 with apt-get by PPA.

This is my Linux distribution.

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.6 LTS
Release:	20.04
Codename:	focal

daniestevez, 2024-09-28:
It might be that the PPA package was built with a different pybind11 (or something else) that what you're using to build gr-satellites. I've seen similar problems in the past. I'll try to replicate the problem in a fresh Docker container and report back.

daniestevez, 2024-09-28:
This is what I've done.

```
docker run --rm -it ubuntu:20.04
apt update
apt install software-properties-common
add-apt-repository ppa:gnuradio/gnuradio-releases
apt install gnuradio
apt install build-essential git gnuradio-dev doxygen libsndfile1-dev liborc-0.4-dev libspdlog-dev python3-construct python3-dev python3-requests python3-six python3-websocket
git clone https://github.com/daniestevez/gr-satellites
mkdir gr-satellites/build
cd gr-satellites/build
apt install cmake
cmake ..
make -j$(nproc)
make test
```

All the tests that involve C++ blocks fail.

I investigated more in the following way.

```
make install
ldconfig
python3
import satellites
```

I got the following error:

```
root@bd6f65f0e0bd:/gr-satellites/build# python3
Python 3.8.10 (default, Sep 11 2024, 16:02:53) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import satellites
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/satellites/__init__.py", line 54, in <module>
    from .adsb_kml import adsb_kml
  File "/usr/local/lib/python3.8/dist-packages/satellites/adsb_kml.py", line 15, in <module>
    from .telemetry import gomx_3 as tlm
  File "/usr/local/lib/python3.8/dist-packages/satellites/telemetry/__init__.py", line 32, in <module>
    from .cute_70cm import cute_70cm
  File "/usr/local/lib/python3.8/dist-packages/satellites/telemetry/cute_70cm.py", line 86, in <module>
    class FswParserState(Enum):
TypeError: function() argument 'code' must be code, not str
```

So I decided to update the version of construct using pip.

```
apt install python3-pip
pip3 install --upgrade construct
```

I now run `make test` again and all the tests pass.

My conclusion is that the only thing to keep in mind to run the current gr-satellites in Ubuntu 20.04 with the GNU Radio 3.10.7 from the PPA is that the version of construct in Ubuntu 20.04 is too old, so a newer version needs to be installed (for instance using pip as I did). I don't know why you got the `AttributeError: module 'satellites' has no attribute 'nrzi_decode'`. I can't replicate it.

daniestevez, 2024-10-25:
This has been opened for 2 months without activity, and I could not replicate the problem. Can the issue be closed?

jane0828, 2024-10-25:
Hi, I upgraded my Ubuntu version and it worked!!
Thanks for helping me.

daniestevez, 2024-10-25:
Great. Thanks! Closing.

closed 2024-10-25 by daniestevez