daniestevez/gr-satellites #679: ASRTU-1 telemetry decoding works in grc3.10 but fails in grc3.8

CLA-179, 2024-10-30:
Using the main branch to decode the telemetry information of ASRTU-1 in grc3.10 can output it normally, but using the same method to decode the same wave file in grc3.8 does not output any results.
The sampling rate I use is 48k.
In both 3.8 and 3.10, it will report:
`gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 39098
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 43475
gr::log :DEBUG: decode_rs1 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 54618
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 111868
gr::log :DEBUG: decode_rs1 - Reed-Solomon decode fail (interleaver path 0)`

CLA-179, 2024-10-30:
The branch used under grc3.8 is asrtu-3.8.

daniestevez, 2024-10-30:
If you're running this on an older system, perhaps you're hitting this VOLK bug: https://github.com/daniestevez/gr-satellites/commit/511c4fcd22c2889c1cc6f6865da2d421d99bb54b

CLA-179, 2024-10-30:
I have tried it on both a virtual machine and a physical machine. The VOLK version on the virtual machine is 2.2, and the version on the physical machine is 2.0. The virtual machine is Ubuntu 18, and the physical machine is Ubuntu 22.

daniestevez, 2024-10-30:
Then it's definitely not the Volk bug I mentioned. Unfortunately I don't have the time/interest to debug problems that affect gr-satellites only under GNU Radio 3.8. This might also be a dead end, because is the bug is not in gr-satellite but in GNU Radio, it won't get fixed, because GNU Radio 3.8 is end-of-life and unmaintained.

CLA-179, 2024-10-31:
Thanks. Maybe we should update our version. haha

closed 2024-10-31 by CLA-179