daniestevez/gr-satellites #683: ZMQ PUB output extra data

jh4xsy, 2024-11-13:
At ASRTU-1 observation #[10554932](https://network.satnogs.org/observations/10554932/), I got this 223 bytes data by using gr-satellies.

```
03 24 F7 F7 01 CB C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0 C0
```
I also sent this to other application "proxy_mmt_gnu" (part of gr-lilacsat) by ZMQ PUB socket, and I got **55**+223 bytes data.
The extra 55 bytes are as follows, containing the strings "smitter", "9k6 BPSK downlink" and "rs_errors".

```
73 6D 69 74 74 65 72 02 00 11 39 6B 36 20 42 50 53
4B 20 64 6F 77 6E 6C 69 6E 6B 07 07 02 00 09 72 73
5F 65 72 72 6F 72 73 03 00 00 00 00 06 0A 00 00 00
00 DF 01 00
```
Is it possilbe to delete these data?

I used GNU Radio 3.8 + gr-satellies with ASRTU-1 support.

daniestevez, 2024-11-13:
I'm copying here something I wrote to someone else by email.

I believe this is a bug in the proxy_mmt_gui application. The ZMQ PUB Message Sink block is the intended way of sending data to this application, since the flowgraphs in gr-lilacsat/examples/ASRTU-1 also use this block. What this block does is to serialize the PDU as a PMT and send that as a ZMQ message:

https://github.com/gnuradio/gnuradio/blob/main/gr-zeromq/lib/pub_msg_sink_impl.cc#L65

The correct way of processing this message in another application is to deserialize it using the PMT library. However, proxy_mmt_gui doesn't do this. It looks like it skips the first 10 bytes of the message and expects to find the PDU bytes after that:

https://github.com/bg2bhc/gr-lilacsat/blob/master/examples/ASRTU-1/proxy_mmt_gui/proxy_mmt_gui.c#L413

It might be true that this is the case for a PDU without metadata, but:

- It is not the case if the PDU has metadata
- It is not robust, since the PMT serialization format could change in a future version of GNU Radio. I don't think it is considered part of the API, so the serialization could change in a minor version without this being considered an API-breaking change.

What happens in this case is that the PDU from gr-satellites contains some metadata (the (transmitter . 9k6 BPSK downlink) (rs_errors . 0) that gets printed in the VERBOSE PDU DEBUG PRINT), and proxy_mmt_gui is incorrectly interpreting these as part of the PDU data (as you found by doing an ASCII decode of the extra bytes).

As a workaround I think you should be able to use the PDU Remove block (actually two copies of it) to remove the metadata before sending it to the ZMQ PUB Message Sink, but this is far from being a good solution.

jh4xsy, 2024-11-14:
Thanks reply.

I understood that the extra data is metadata.

The gr-satellites's metadata size varies depending on the number of rs_errors. But proxy_mmmt_gui's metadata size is fixed: hard coded. 

Humm....  I agree  proxy_mmt_gui should fix this.