daniestevez/gr-satellites #692: CCSDS Concatenated Deframer causes crash

SJ-Innovation, 2024-12-12:
Hi everyone, 
The CCSDS Concatenated Deframer block causes a crash on startup of the flowgraph. No useful error is returned, only the error 3221225477, segmentation fault. The Reed-Solomon & Uncoded deframers do not cause the same crash, though obviously don't work for the satellite I'm trying to work with.

System: 
GNURadio 3.10.11.0 (Py 3.12.5) installed with Ryanvolz/RadioConda onto Windows 10. 
gr-satellites 5.6.0

I'm just trying a build-from-source of the Git latest, to see if that helps. I'll update with results of that.

daniestevez, 2024-12-12:
Can you either share a small flowgraph that reproduces the problem (does it happen deterministically for some particular inputs?) and/or a gdb stack trace of the segfault?

daniestevez, 2024-12-12:
Also, the CCSDS Concatenated Deframer block is a hierarchical block composed of several blocks, each of which runs in its own thread. So the first thing to find out is what thread (and hence which block) is producing the segfault.

SJ-Innovation, 2024-12-12:
The flow for causing the error is literally just a constant source, the CCSDS Concatenated Deframer & a message debug. Even with default settings the deframer crashes. I've compiled from source, and its the same issue. I'll dig into the hier and narrow down which thread is crashing us. 
[image]

SJ-Innovation, 2024-12-12:
Okay so the FEC Extended Decoder (from which the NASA V D hiers) crashes whenever the decoder object is set to an instance of CC decoder definition. Other decoder definitions do not cause the same crash.

daniestevez, 2024-12-12:
Thanks for the debugging. This doesn't crash on my system (Arch Linux), so it seems like a Windows-specific problem. The FEC Extender Decoder and CC Decoder Definition blocks are in-tree GNU Radio blocks, so it would be good to open an issue in the [GNU Radio repository](https://github.com/gnuradio/gnuradio).

To me, this probably looks like bug in Volk. Perhaps using the env variable `VOLK_GENERIC=1` manages to work around this problem.

closed 2024-12-12 by SJ-Innovation