daniestevez/gr-satellites #735: Please release v5.7.0 to the Ubuntu PPA

DeflateAwning, 2025-04-25:


daniestevez, 2025-04-25:
gr-satellites v5.7.0 is available in the Ubuntu PPA for Ubuntu 24.04 and 24.10. The policy that I follow is to make a package for the latest Ubuntu LTS and latest Ubuntu at the time when the gr-satellites version is released. I have sometimes also made a package of the latest gr-satellites release just after a new Ubuntu LTS has come out.

At this moment, there is no gr-satellites package in the PPA for Ubuntu 25.04, since no gr-satellites version has been released since 25.04 has come out, and 25.04 is not an LTS.

This policy is what I believe is a fair use of my time, since at the moment publishing a package to the PPA is a manual and somewhat time consuming process. If someone from the community (maybe you?) is willing to take over this task and provide packages for more Ubuntu versions, I would be happy to pass on the task.

daniestevez, 2025-04-28:
Yes.

daniestevez, 2025-05-18:
Closing as not planned.

closed 2025-05-18 by daniestevez