daniestevez/gr-satellites #753: "Could not find port" error

PratikAher20, 2025-06-01:
Hi everyone, 
Recently while I was trying to demodulate and decode a pre recorded wav file, I came across this error stating "Could not find port pdus in". On further debugging the issue I found that this error only happens to occur while using the CCSDS RS Deframer block. 
I have attached a screenshot of the flowgraph and the error message. 
Can someone suggest any solution for this problem. 

Versions:
gr_satellite: v5.5
gnuradio: v3.10

[image]

Regards, 
Pratik Aher.

daniestevez, 2025-06-01:
Also potentially relevant:
- What distribution are you using and how have you installed GNU Radio and gr-satellites?

daniestevez, 2025-06-02:
You don't give all the details I asked for, but be aware that if by any chance you have installed GNU Radio from the Ubuntu PPA (the info you give doesn't show whether this is true or not) and gr-satellites from the Ubuntu PPA (you have done that according to one of your screenshots), then you'll get random segfaults like this. The GNU Radio PPA and gr-satellites PPA are not compatible.

PratikAher20, 2025-06-03:
> random

[image]

I have attached a zip file containing the recording and the gnuradio flowgraph.

PratikAher20, 2025-06-05:
I did installed the complete source code and I am able to run the flowgraph completely without any warnings and segmentation fault error. 
Thanks @daniestevez !