daniestevez/gr-satellites #756: Cannot decode anything

ik8ozv, 2025-08-27:
Hello,
I would like to know if someone can help me.
Unfortunately I was off the satellites from last year, due to problems with my
Elevation Motor. Now seems to be OK and I would like to restart again but I got problem
To receive satellites with Ubuntu. I do not get any incoming data either from both Gnuradio and Terminal, viceversa if i record the audio and then i pass it in the gnuradio decoder, i got incoming data.
Please any help?
Thanks in advance

73' s de Enzo IK8OZV

ik8ozv, 2025-09-13:
Jan what irnformation you need?
I am using GQRX to receive and also to record
Please let me know what you need to know to gelp.
Thanks

73's de Enzo IK8OZV

daniestevez, 2025-09-13:
I think I or others have said this before, but I will say it again. Unless you explain your problem in a way that allows other people to reproduce it (this includes describing all the steps that you are doing and sharing recordings so that people can use the same input), it is very difficult to give any help and you are unlikely to get any replies in this issue.

closed 2025-11-08 by daniestevez