daniestevez/gr-satellites #757: Build from source error: ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"

janvgils, 2025-09-13:
It was time to execute a new build from source on my Debian Forky system.
All seems to be fine until the `ax100_decoder` error.

Looking at git log, the latest commit from this source is `commit 92737bd25bdb338274b09c1eae7059d361b82577 (HEAD -> main, origin/main, origin/HEAD)`

I tried to build this source on two different systems running Debian Forky and both have this problem, a Debian Trixie is building without problems.

`gr_satellites --version`

```
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_usp_ax25_crop.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_usp_pls_crop.block.yml
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/dist-packages/satellites/__init__.py", line 47, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 21, in <module>
    import satellites.core
  File "/usr/local/lib/python3.13/dist-packages/satellites/__init__.py", line 49, in <module>
    from .satellites_python import *
ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"
```

`make test`

```
Running tests...
Test project /home/janvg/source/git/daniestevez/gr-satellites/build
      Start  1: qa_costas_loop_8apsk_cc
 1/28 Test  #1: qa_costas_loop_8apsk_cc ..........***Failed    0.19 sec
      Start  2: qa_crc
 2/28 Test  #2: qa_crc ...........................***Failed    0.18 sec
      Start  3: qa_fixedlen_tagger
 3/28 Test  #3: qa_fixedlen_tagger ...............***Failed    0.18 sec
      Start  4: qa_fixedlen_to_pdu
 4/28 Test  #4: qa_fixedlen_to_pdu ...............***Failed    0.18 sec
      Start  5: qa_hdlc
 5/28 Test  #5: qa_hdlc ..........................***Failed    0.18 sec
      Start  6: qa_kiss
 6/28 Test  #6: qa_kiss ..........................***Failed    0.18 sec
      Start  7: qa_manchester_sync
 7/28 Test  #7: qa_manchester_sync ...............***Failed    0.18 sec
      Start  8: qa_mobitex_fec
 8/28 Test  #8: qa_mobitex_fec ...................***Failed    0.18 sec
      Start  9: qa_mobitex_fec_block
 9/28 Test  #9: qa_mobitex_fec_block .............***Failed    0.18 sec
      Start 10: qa_mobitex_to_datablocks
10/28 Test #10: qa_mobitex_to_datablocks .........***Failed    0.18 sec
      Start 11: qa_nrzi
11/28 Test #11: qa_nrzi ..........................***Failed    0.17 sec
      Start 12: qa_pdu_add_meta
12/28 Test #12: qa_pdu_add_meta ..................***Failed    0.18 sec
      Start 13: qa_pdu_head_tail
13/28 Test #13: qa_pdu_head_tail .................***Failed    0.18 sec
      Start 14: qa_pdu_length_filter
14/28 Test #14: qa_pdu_length_filter .............***Failed    0.18 sec
      Start 15: qa_pdu_scrambler
15/28 Test #15: qa_pdu_scrambler .................***Failed    0.17 sec
      Start 16: qa_phase_unwrap
16/28 Test #16: qa_phase_unwrap ..................***Failed    0.18 sec
      Start 17: qa_rs
17/28 Test #17: qa_rs ............................***Failed    0.17 sec
      Start 18: qa_selector
18/28 Test #18: qa_selector ......................   Passed    0.23 sec
      Start 19: qa_viterbi
19/28 Test #19: qa_viterbi .......................***Failed    0.17 sec
      Start 20: qa_ao40_fec_deframer
20/28 Test #20: qa_ao40_fec_deframer .............***Failed    0.18 sec
      Start 21: qa_mobitex_deframer
21/28 Test #21: qa_mobitex_deframer ..............***Failed    0.17 sec
      Start 22: qa_usp_deframer
22/28 Test #22: qa_usp_deframer ..................***Failed    0.17 sec
      Start 23: qa_kiss_server_sink
23/28 Test #23: qa_kiss_server_sink ..............***Failed    0.18 sec
      Start 24: qa_telemetry_parser
24/28 Test #24: qa_telemetry_parser ..............***Failed    0.18 sec
      Start 25: qa_rms_agc
25/28 Test #25: qa_rms_agc .......................***Failed    0.17 sec
      Start 26: qa_rms_agc_f
26/28 Test #26: qa_rms_agc_f .....................***Failed    0.17 sec
      Start 27: qa_satyaml
27/28 Test #27: qa_satyaml .......................   Passed    0.62 sec
      Start 28: qa_by02
28/28 Test #28: qa_by02 ..........................***Failed    0.17 sec

7% tests passed, 26 tests failed out of 28

Total Test time (real) =   5.45 sec

The following tests FAILED:
	  1 - qa_costas_loop_8apsk_cc (Failed)
	  2 - qa_crc (Failed)
	  3 - qa_fixedlen_tagger (Failed)
	  4 - qa_fixedlen_to_pdu (Failed)
	  5 - qa_hdlc (Failed)
	  6 - qa_kiss (Failed)
	  7 - qa_manchester_sync (Failed)
	  8 - qa_mobitex_fec (Failed)
	  9 - qa_mobitex_fec_block (Failed)
	 10 - qa_mobitex_to_datablocks (Failed)
	 11 - qa_nrzi (Failed)
	 12 - qa_pdu_add_meta (Failed)
	 13 - qa_pdu_head_tail (Failed)
	 14 - qa_pdu_length_filter (Failed)
	 15 - qa_pdu_scrambler (Failed)
	 16 - qa_phase_unwrap (Failed)
	 17 - qa_rs (Failed)
	 19 - qa_viterbi (Failed)
	 20 - qa_ao40_fec_deframer (Failed)
	 21 - qa_mobitex_deframer (Failed)
	 22 - qa_usp_deframer (Failed)
	 23 - qa_kiss_server_sink (Failed)
	 24 - qa_telemetry_parser (Failed)
	 25 - qa_rms_agc (Failed)
	 26 - qa_rms_agc_f (Failed)
	 28 - qa_by02 (Failed)
Errors while running CTest
Output from these tests are in: /home/janvg/source/git/daniestevez/gr-satellites/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
make: *** [Makefile:71: test] Error 8
```

I can imagine that more information is needed, a full build log with version and env information is available via the following link.

Jan | PE0SAT

daniestevez, 2025-09-13:
Thanks for reporting. This looks the same as #634. If you are not able to build and use other out-of-tree modules either, then this is definitely a problem with Debian packaging. The best person to reach out would be Maitland Bottoms. He is the Debian GNU Radio packager, and usually hangs around the GNU Radio Matrix chat (for instance in the ham radio room).

janvgils, 2025-10-23:
After today's Debian Forky gnurradio-dev update to:

```
dpkg -l | grep gnuradio-dev
ii  gnuradio-dev                                  3.10.12.0-5                              amd64        GNU Software Defined Radio toolkit development
```

gr-satellites can be build again:

```
Copyright 2016-2025 Daniel Estevez

License GPL-3.0-or-later: GNU GPL version 3 or later
<https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Thanks for the support